opendata-stuttgart/sensors-software #419: Duel Particle  sensors

jordanc365, 2019-06-19:
Hello,
Is it possible to equip the platform with 2 dust sensors instead of a temp/ humidity sensor?
It could be helpful as one could see that one of the sensors are degrading if they report different values.
Here's an example of a sensor that does this: 
[https://www2.purpleair.com/products/purpleair-pa-ii-sd](url)

ricki-z, 2019-06-19:
This would be a very instable setup as the NodeMCU isn't really fast enough to simulate 2 serial interfaces at the same time. And there is no hardware serial which we could use.

Adorfer, 2019-06-20:
for dual RS232 it would be at least neccesary to pass from ESP8266 to the more powerfull ESP32 (which is essential at least 3€ more expensive)

jordanc365, 2019-07-12:
Thanks for the help! I'll have a look into it

closed 2019-07-12 by jordanc365