opendata-stuttgart/sensors-software #420: Definition of measurement constants

jordanc365, 2019-06-24:
Can somebody please explain what the following measurement constants are in the code?

const unsigned long SAMPLETIME_MS = 30000;
const unsigned long SAMPLETIME_SDS_MS = 1000;
const unsigned long WARMUPTIME_SDS_MS = 15000;
const unsigned long READINGTIME_SDS_MS = 5000;
const unsigned long SAMPLETIME_GPS_MS = 50;
const unsigned long DISPLAY_UPDATE_INTERVAL_MS = 5000;

ricki-z, 2019-06-24:
const unsigned long SAMPLETIME_MS = 30000;
minimum time span between two measurements for the PPD42NS (the first PM sensor we tested).

const unsigned long SAMPLETIME_SDS_MS = 1000;
minimum time span between two measurements for the SDS011.

const unsigned long WARMUPTIME_SDS_MS = 15000;
minimum time of running before the first value is read for the SDS011.

const unsigned long READINGTIME_SDS_MS = 5000;
how long values are read from the SDS011 (this will give 4-5 values, for the transmission the average of these values is used)

const unsigned long SAMPLETIME_GPS_MS = 50;
time between two samples read from the GPS sensor (GPS sensor would fill up the serial buffer if we don't read this often enough)

const unsigned long DISPLAY_UPDATE_INTERVAL_MS = 5000;
time between switches of displayed screen (for all display types)

jordanc365, 2019-06-28:
Hey @ricki-z 
How would i go about setting the sensor to record constantaly outputting minute averages?
No matter what combinations I try I can't seem to get it to work...

ricki-z, 2019-06-28:
@jordanc365 this should work with READINGTIME_SDS_MS = 60000
and a measurement interval of 60 seconds. And you should disable Auto_Updates as this would install the default firmware with the default READINGTIME_SDS_MS of 5000.

ricki-z, 2019-06-28:
And in this case the live time of the sensor will be only 8000 hours (slightly less than one year).

jordanc365, 2019-07-12:
Thanks @ricki-z your help is much appreciated!
I found that setting the measurement interval to 60.01 seconds works well :) 
I think there needs to be a better explanation of these variables in the code.
The measurement interval needs to be greater than the sum of the reading time and the warmup time :. the default code crashes if you try to sample in intervals <20 seconds. 
This is probably super obvious to everyone but as someone new to the project it was a lightbulb moment! :)

closed 2019-07-12 by jordanc365