opendata-stuttgart/sensors-software #421: No data from Sensor...

ansim58, 2019-06-26:
Hi,
since 1 day i get only Temperature and humidity data.
So I want to check the SDS011.
Here are my questions: How can I read the debug information? What is the effect of setting the debug level to Error, Warning, Info? Where can I read the logs?
Thank you very much and best regards, Andreas

EdgarMCR, 2019-06-26:
Have you restarted it?  (Switched off power, then on again)

My sds011 stops about once a month like this but restarting it fixes the problem

Bouni, 2019-06-27:
Same problem here for me. Built a sensor for our Hackerspace about a week ago and it seemed to work just fine but has gaps for particle measurements while temperature and humidity working fine.

[image]

Restarted the sensor yesterday but worked only for hours.

ansim58, 2019-06-27:
Thanks to Edgar!
After pulling the plug everything is working fine again.
What I did before was only a soft-restart, which didn't work.
Sometimes things aren't as complicated as we think... ;-)

ricki-z, 2019-06-28:
@ansim: would only answer the question about debug levels and output.
We can only send the debug messages over the serial interface of the NodeMCU. So you have to connect the NodeMCU to a computer where you can see the output either with the Arduino IDE or any serial monitor.
The levels should be self explaining. The lowest level Error will show errors only (sensor that couldn't be initiatet, communication errors), Info will also show the steps the firmware is going through (sensor reading, sending data, sending web page content, ... ). At max. level you can also see the single bytes that are read from the PM sensors. But this may produce a high volume of output.

ricki-z, 2019-06-28:
@Bouni Have you checked the cables? And you could try to replace the USB wall charger. Some of them don't have 5V output. If the Output is too low the SDS011 may not work.

dirkmueller, 2019-11-04:
@ansim58 any outcome of this? can this be closed? any improvement to make in the software?

Bouni, 2019-11-05:
Hi everyone, 
for me the sensor works now but not as I planed. I powered it in the first place directly via the VIN and GND pins on the board with 5VDC from a Stepdown converter that generates 5VDC from 24VDC (up to 3A). 
That didn't work reliably so I switched to a USB powersupply and now the Sensor seems to work.
I've monted it indoors a while ago to see if the problem is solved, but in the long run I want to have it outdoors and don't want to run a long USB cable to the outside, so I defenitely need to find the reason for the direct power issue.

closed 2019-11-05 by ansim58