opendata-stuttgart/sensors-software #422: Dew Point

snookie, 2019-06-29:
Hi,

it would be nice to get the Dew Point too. If you have the Tempreture and Humidity, you can calculate the Dew Point.  

The Fomular (here in JS): https://myscope.net/taupunkttemperatur/

Adorfer, 2019-06-29:
Since the design of the network is, that all calculation ("correction) is done in the backend, i do not see a way to get the real humidity from what DHT22 (or BME180) reports, since not only the absolute values differ, but as well the "form of curves".

ricki-z, 2019-06-30:
The values of the DHT22 are not really reliable enough to compute a dew point. And as mentioned this should be done on the backend or API client. The dew point could be calculated at any time (if you have correct measurements), so we don't need to save or transmit it. And we have to save space in our firmware (at the moment we use 94% of the available space).