opendata-stuttgart/sensors-software #423: System crashes during WiFi networks visualization on web config

executer-uno, 2019-07-07:
Hi, Im now manage to get readings from PMS7003 dust sensor with BME280
https://easyeda.com/mineshoppingcart/dust-sensor - schematics. For now only first PMS7003 sensor used.

Issue conditions: empty WiFi config or WiFi network not available. Sensor establishes its own acess point for initial config. It waits until browser connects to it (laptop Chrome or Android phone chrome tested). Then it crushes right after WiFi networks detection (count of networks is correct):

> 10:09:22.569 -> output config page ...
10:09:22.569 -> output config page 1
10:09:23.271 -> wifi networks found: **2**
10:09:23.271 -> output config page 2
10:09:23.318 -> **Fatal exception 28(LoadProhibitedCause):**
10:09:23.364 -> epc1=0x4020aca3, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000054, depc=0x00000000
10:09:23.458 -> 
10:09:23.458 -> Exception (28):
10:09:23.458 -> epc1=0x4020aca3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000054 depc=0x00000000

So now I can work with it if I define correct WiFi settings in firmware from arduino IDE.

Full log and browser pre-crush screenshot are in attachements.
Compiled under Arduino 1.8.8
Libs screenshot is in attachement.

[image]

[image]

[LOG.txt](https://github.com/opendata-stuttgart/sensors-software/files/3365369/LOG.txt)

ricki-z, 2019-07-07:
There is a memory bug in the beta branch. We haven't found the source of this crash until now.
The master branch should work without that problem.

ricki-z, 2019-07-07:
And another comment to your project.
Using two software UARTs may cause crashes. The SoftwareSerial lib is working with interrupts but the ESP8266 sometimes can't handle them fast enough for 2 UARTs.

dirkmueller, 2019-10-20:
This is a problem with the master branch. in beta it was fixed in d461dd23d050fd3d30acc220d53ce20cb72edc27 - so the next stable release should have this fixed.

dirkmueller, 2019-11-04:
@executer-uno can you confirm this is fixed in the latest firmware for you?

dirkmueller, 2019-11-26:
@executer-uno any feedback? Would like to close the issue.

dirkmueller, 2019-11-27:
Please reopen if there is still an issue. Thanks!

closed 2019-11-27 by dirkmueller