opendata-stuttgart/sensors-software #428: Sending data to InfluxDB not working

ca057, 2019-07-17:
Hello!

I want to send the data from the sensor to an InfluxDB instance. Unfortunately I seem not to be able to configure the sensor in a way that the data reaches my InfluxDB-instance.

My setup is the following:
I have nginx running, which listens on port 443 and is configured with a self-signed certificate created with 2048 bit. All requests to are forwarded to the InfluxDB-instance running in the background.

On the sensor, I configured the connection to InfluxDB:
```
Server: <SERVER_IP_ADDRESS>
Path: /write?...
Port: 443
User: <MY_USERNAME>
Password: <PASSWORD>
```

When I use the same parameters to perform the requests via Postman, the data gets successfully written into InfluxDB. When I restart the sensor with the mentioned configuration, I never get any requests to my server. The access logs of nginx also don’t show any incoming requests.

Did I miss something?

Thanks a lot for your help!

ricki-z, 2019-07-17:
Can you send me a screenshot of th InfluxDB part on the config page?
Email address is rajko (at) luftdaten.info

ricki-z, 2019-11-13:
@ca057 is this issue solved and we can close it?

ca057, 2019-11-13:
> @ca057 is this issue solved and we can close it?

With a new setup it’s working, it seemed to have been a mismatch in which ciphers the server is supporting.

Thanks!

closed 2019-11-13 by ca057