opendata-stuttgart/sensors-software #429: ESP8266 requires manual reset after power loss

jordanc365, 2019-07-22:
I'm powering my unit using solar and a massive battery and for the most part it works pretty well but if the battery runs low even for a few minutes and the Nodemcu restarts it's stuck in a state in which it will not measure or transmit data. 
It's simialr to the issues identified in these posts:
[Here](https://github.com/esp8266/Arduino/issues/1074
[Here](https://stackoverflow.com/questions/46367463/nodemcu-arduino-ide-after-power-off-sketch-doesnt-start
[Here](https://arduino.stackexchange.com/questions/36938/nodemcu-does-not-boot-up-when-gpio-pin-is-high)
[Here](https://forum.arduino.cc/index.php?topic=542780.0)
Does anyone have experience with issues like this?
Cheers!

dirkmueller, 2019-11-27:
I am afraid there is not a whole lot we can do about this.. usually the mcu hangs before even the initial setup function is being invoked. Esp8266 is particularly power hungry during early start ..

Agree to close? You can tey finding help on chat.luftdaten.info but this is strictly a bug tracker foe the firmware..

DeeKey, 2019-12-18:
Think - it should go to FAQ section. For such autonomous systems HW reset is required. Also, maybe even for standalone systems HW reset based on timer is a nice idea if you want constant data.