opendata-stuttgart/sensors-software #430: How to use the "Send to CSV" option (to use the sensor as a mobile sensor)

MagTun, 2019-07-23:
I sometimes would like to use the sensor as a mobile one, to check the PM in the other streets around my neighborhood.
I will be out of reach of my wifi so I need a local storage of the PM value so that I can retrieve them later. I have seen the option "Send to CSV" in the config and I also read [that we can output the CSV via USB](https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/Readme.md):
> The data can be output as CSV via USB. For this purpose, debug should be set to 0 both in ext_def.h and in the WLAN configuration so that the data output is only the sensor data. When restarting the ESP8266 then only a few characters appear, which represent the start state. 

But I don't understand how it works. So here are my questions:
- If I check the box "Send to CSV", then "Save and restart", then I go in out of reach of my wifi, will the data be stored in the CSV right away or will there be a delay (like 5 min) during which the sensor will try to connect to the wifi or access point and only after this delay, it will store the value in the CSV? 
- I have seen that I can set debug to 0 in Config/More settings but I don't understand how I can set the debug to 0 in ext_def.h. Will checking "Send to CSV" do that for me?
- How can I retrieve the CSV by USB? Should I connect the esp8266 or the SDS011 to the USB port of my computer?

ricki-z, 2019-07-24:
The data will be "sent" as CSV to the USB port. So you need to connect the NodeMCU to a computer while measuring.
"Send to CSV" won't set the debug level to zero. You have to do this via the config page.
In ext_def.h you only need to replace the "3" behind DEBUG with "0", but this setting is overwritten with the value set in the saved configuration.
If there is no wifi connection available the sensor will start the config mode (acting as an access point). This mode is open for 10 minutes before the sonsor will close the access point and start measuring. But this can be shortened by setting the "Duration router mode" to a smaller value.

MagTun, 2019-07-24:
1) Just to be sure, setting debug to 0 in the config page will set 0 in ext_def.h? So each time I want to "send to CSV", I only need to make sure that Debug level in the config page is set to 0? Or should I also click on "Debug Null" on  the home page (I guess it's the same setting as Debug level on the config page) 

2) Is it possible to set the duration router mode to 0 or 1 (so it won't waste energy acting as an access point when I am out of my wifi range)? If I understand well, this means that by setting a value so low I won't anymore have the time to connect to the "access point config page" where we normally give the Wifi name/password to the sensor so the NodeMCU can connect to the internet. This means that while my sensor is out of range, I shouldn't change my Wifi name/password on my router otherwise the NodeMCU won't be able to reaccess the config page. If this happens (if duration router mode is set to 1sec and I changed my wifi password), how can I give back to my sensor the access to my wifi? Should I reinstall the firmware on my ESP-8266 from scratch ?

ricki-z, 2019-07-25:
ext_def.h is setting the default values for compiling. We use this to generate different precompiled firmware binaries i.e. for users with a DHT22 or a BME280. These settings are overwritten by the settings in the local sensor configuration (which is saved as a file in the SPIFFS part of the flash).
So flashing a new firmware doesn't change the active configuration. So setting the debug level to 0 on the config page is enough. You don't need to change the ext_def.h.
The "Debug level" buttons on the web page are for development or debugging. They change the debug level only temporary without changing the configuration.
Setting the router mode duration to 0 or 1 would effectively disable the router mode. So you won't be able to configure the sensor if it can't access a wifi network.
Do "reset" the configuration (by deleting the config file) you need to flash an empty SPIFFS file system. The instructions for this can be found at https://github.com/opendata-stuttgart/meta/wiki/EN-Installing-firmware#deleting-the-configuration . The next version of our flashing tool will have an option to delete the configuration but at the moment we have some minor problems with compiling this tool for all platforms.

MagTun, 2019-07-25:
Thanks so much ricki-z for taking the time to write such a detailed answer!
A lot of things make sense now that I understand that the config settings are saved in a SPIFFS file that is separate from the firmware. 

This flashing tool is simply awesome. That's a great improvement from the previous process with Arduino IDE and esp-tool and all that.

ricki-z, 2019-07-26:
Your thanks for the flashing tool should go to @Informatic. He mainly developed this tool.
(This message should forward this to him ;-) )