opendata-stuttgart/sensors-software #431: Support for the SPI/I2C version of the SSD1306 (7 pins)

MagTun, 2019-07-25:
I have seen that the luftdaten sensor can support several displays like  LCD 1602 (I2C: 0x3F and 0x27),  OLED SH1106 and OLED SSD1306. But apparently it doesn't support the SPI/I2C version of the SSD1306.

This version of the SSD1306 has 7 pins and support both I2C and SPI (from [banggood](https://www.banggood.com/fr/7Pin-0_96-Inch-OLED-Display-12864-SSD1306-SPI-IIC-Serial-LCD-Screen-Module-For-Arduino-p-1364267.html?rmmds=myorder&cur_warehouse=CN) and from an [instructable](https://www.instructables.com/id/Wemos-D1-Mini-096-SSD1306-OLED-Display-Using-SPI/)):
GND (Ground)
VCC (3.3 or 5V DC power)
D0 (Clock)
D1 (Data: SDA or MOSI)
RES (Reset)
DC (Data/Command)
CS (Chip Select)

I have tried connecting it with just 4 pins like a regular I2C connection but it doesn't work:

SSD1306  (SPI) | --- | NodeMCU
--- | --- | ---
VCC | -> | Pin 3V3
GND | -> | Pin GND
D0 (SCL) | -> | Pin D4 (GPIO2)
D1 (SDA) | -> | Pin D3 (GPIO0)

Am I making a mistake?

ricki-z, 2019-07-26:
This display version needs a reset pin. But as we don't have any free pins left we can't support this display version. The firmware only supports the "pure" I2C displays where the reset is done by a I2C command.

MagTun, 2019-07-26:
This makes sense! Thanks!

closed 2019-07-26 by MagTun