opendata-stuttgart/sensors-software #432: Humidity went down to 1% over 3hrs and stays there. DHT22 defect?

reibuehl, 2019-07-30:
This morning, the humidity readings from the DHT22 on my sensor went gradually down from ~40% to 1% over a period of ~3hrs and are now stuck at 1%. The humidity sensor of my weather station at the same location reports 35-40% the whole time. The temperature readings from the DHT22 and the weather station are in sync, so that part of the DHT22 seems to work. 

Is this a common sign of a defective sensor or is there a way to "revive" the DHT22?

stewecar, 2019-07-30:
there is a way to revive it..it is also on the sensor's datasheet at the end of it..
it's a kind of recalibration..really not very easy but you can find something online.
cheers.

reibuehl, 2019-08-01:
I found section 8, point 5 in this one:
https://akizukidenshi.com/download/ds/aosong/AM2302.pdf
Is that the procedure you had in mind? That is really well beyond what I can do. I was more looking for something like a software reset or so...

stewecar, 2019-08-01:
That's what I meant..
I don't think there's a way to reset It..only that recalibration procedure Is written..
Consider that is a really cheap sensor..
Cheers.

reibuehl, 2019-08-01:
I just saw that BME280 board is just over 5 EUR, so I will replace the DHT22 with a BME280 which should be better and adds pressure measurement too.

closed 2019-08-01 by reibuehl