opendata-stuttgart/sensors-software #433: Add information to field lengths and correct current field lengths

ss89, 2019-08-06:
A friend of mine bought some sensors and uses it with this firmware.
I did setup some influxdb for him to use, ofcourse with the maximum length i thought possible for this firmware, which looked like it would be 50 characters.
Since this is the maxlength in the user interface.
However it seems that due to null termination you can only use 49 characters(?).

Feature request:
* Please add the length of the fields in brackets next to the fields label
* Please fix the max-length attribute in the html so that it reflects the real maximum

ricki-z, 2019-08-07:
The fields host and path for the InfluxDB config are limited to 100 byte (minus 1 for null termination). This should be enougth for most cases. The fields should be saved as 1 byte per character. If you use Unicode chars the useable length is only 50 chars.

ss89, 2019-08-12:
It seems that i looked at a fork or so. 
Never the less, the interesting parts username and password are 65 bytes in size.
Or my friend used an outdated version with lower field lengths?
Still wondering why it did only work with a lower length for the password field.

JDTm, 2019-08-12:
Installed Firmware: NRZ-2018-123B

dirkmueller, 2019-11-10:
The max-length attribute has been fixed already in the beta branch. I can add the max input length as an hover overlay, however showing it always looks far too intrusive for such a (relatively unimportant) piece of information.

dirkmueller, 2019-11-27:
This has been implemented in beta branch. Please don't hesitate to reopen if there is still an issue remaining.

commit 5de2d7f4 (2019-11-11): Merge pull request #545 from dirkmueller/beta

Add hover titletip for maximum field lengths (Fixes #433)
commit fcfc4764 (2019-11-11): Add hover titletip for maximum field lengths (Fixes #433)

This might be a bit awkward, but I think showing it generally
looks even worse. as a hover tooltip this seems to be fine however.
closed 2019-11-27 by dirkmueller