opendata-stuttgart/sensors-software #439: BME280 keine Daten mehr nach ca. 1-2 Tagen

FreeTechNick, 2019-08-27:


dirkmueller, 2019-09-03:
@FreeTechNick ich hatte die gleichen Symptome, bei mir lag es an stoerungen auf dem I2C bus (verursacht wohl durch feuchtigkeit am sensor). ein einspruehen des sensors mit All-Wetter-Lack (vorsicht !!!! sensor vorher sicher abdecken damit das luftloch nicht mit engesprueht wird) hat geholfen.

Die Software muesste fehlertoleranter gegenueber Stoerungen auf dem i2c bus gemacht werden..

brandungskieker, 2019-09-13:
Habe das gleiche Symptom und auch schon mal in https://github.com/opendata-stuttgart/sensors-software/issues/386 beschrieben.

brandungskieker, 2019-09-13:
> @FreeTechNick ich hatte die gleichen Symptome, bei mir lag es an stoerungen auf dem I2C bus (verursacht wohl durch feuchtigkeit am sensor). ein einspruehen des sensors mit All-Wetter-Lack (vorsicht [...]

Hat die Feuchtigkeit den Sensor bei dir zerstört oder hat der Reboot ebenfalls geholfen?

dirkmueller, 2019-09-13:
Reboot hilft kurzfristig, aber in meinem fall gab es ausgeprägte Korrosion auf der Platine. Mittels Zahnbürste abschrubben war das zu beheben und der Sensor lief dann wieder.

Ich habe aber die 4 euro investiert und einen neuen gekauft, den mit harzlack wasserfest gemacht und seitdem keine Probleme mehr.

dirkmueller, 2019-11-27:
Gibt's hier Neuigkeiten?

MagTun, 2020-01-08:
Same here I had a BME280 which worked for a couple of days (weeks?) and then it stopped sending data. It had some corrosion, which I removed with a toothbrush. I applied some nail polish and since then I don't have any bug.

DeeKey, 2020-01-15:
Seems that this is not software bug. Please close if the issue is solved!