opendata-stuttgart/sensors-software #440: sensorSDS() reads stale bytes

chocotov1, 2019-08-27:
Disclaimer: I haven't tested or noticed any real world problem. Maybe i'm wrong, then my bad :)

I'm in the process of making a Lora based (Arduino UNO / ATtiny, battery / solar powered) node of the sds011 data. I was interested how the bytes of the sds011 were read in Luftdaten.

When looking at sensorSDS() in airrohr-firmware.ino I was surprised that this seems to work:

while (serialSDS.available() > 0)

When I made my version a couple of days ago, reading the serial was quicker than the sensor was transmitting. I added a short timeout while reading the serial. It also makes sense to do that, because the sensor only sends once per second. The timeout needs to be at least one second.

I'm speculating this wasn't an issue here because:
- serialSDS() is slow in processing the bytes
- (more likely) the buffer of the serial contains previous data

Im not sure how big the buffer of this software serial is. Personally i'd just wait for the next data. My version at the moment settles for the average of the last four readings.

chocotov1, 2019-08-28:
> Im not sure how big the buffer of this software serial is. 

I guess it's 128 bytes.

`SoftwareSerial serialSDS(PM_SERIAL_RX, PM_SERIAL_TX, false, 128);`

The data of the SDS011 is 10 bytes long. My guesstimate is that the bytes read by sensorSDS() could be up to 12 SDS011 updates old.

ricki-z, 2019-08-28:
One cylce of the main loop is about 10 milliseconds. And while the sensor is active we read all available bytes from the serial buffer.
With 9600 Baud the 10 bytes should need slightly more than 1 millisecond. And in active mode only 1 transmission per second needs to be read.
So there should be enough time that all bytes are written to the buffer before we read the bytes. the serial interface is emulated in software. So while data is transmitted your application should run slightly slower.
If you want to make sure that a complete transmission is available in the buffer you should set the condition to:
`while (serialSDS.available() > 19)`

chocotov1, 2019-08-29:
I didn't notice the function was called multiple times. But if it runs so often that the buffer becomes empty then my alleged scenario of this issue doesn't occur.

Thanks for taking the time!

closed 2019-08-29 by chocotov1