opendata-stuttgart/sensors-software #441: Features request: in the configuration screen add an option (1) to rotate the screen and (2) to not display wifi info and device info

MagTun, 2019-08-30:
It would be great if in the configuration page we could choose to:
- rotate the screen by 180° ([someone else ](https://github.com/opendata-stuttgart/sensors-software/issues/389)had the same problem than me). When the checkbox in the configuration is checked, then the setup will apply this: `display.flipscreenvertically()` 

- not to display the Wifi info and Device info: when I use the screen, I would like to be able to read the PM value all the time, and not to wait for the Wifi info page and the Device info page to be displayed and then get a quick chance to read the PM value before having to wait again. Here again, 2 checkboxes in the configuration page would allow the user to choose what the display shows. This option is essential for sensors used as mobile. 

Also, it would be great if the sensor doesn't apply the measuring interval for the very first measurement. Let say, my measuring interval is 60 sec then, with the current setup, I need to first wait 60 sec before the first measurement then I get my first measurement, then I wait again 60 sec before getting the second measurement.   
But if the sensor doesn't apply the measuring interval for the very first measurement, then I can have 2 measurements in almost 60 secs.  This is also essential for sensors used as mobile.

ricki-z, 2019-08-31:
The flip option is available in the actual beta now and will be available in the next release.
There is only limited RAM so the config file may become too large if we add too many options.
So is that really a problem to wait the 10 seconds for the 2 screens with the wifi and device infos?

To the first time displaying values: The PM sensors need a warmup time. So the first measurement won't be available earlier than 20 seconds after startup (15 seconds warmup + 5 seconds measurement).
And this firmware wasn't really developed for mobile usage and short measurement intervals. For mobile usage you should think about a solution with i.e. a Raspberry Pi where you can save the measurement locally and that has a cpu fast enough to handle the SDS011 and a GPS modules at the same time.

MagTun, 2019-09-03:
1) That's great for the flip option! I hope the next release comes soon!

2) Yes it really is a problem to wait 10 seconds for the 2 screens with the wifi and device infos: imagine you have a watch and want to know the time but each time you look you have  10 sec ads. 

3) Then maybe the first measurement can be 20 secs after startup and after that the measurement intervals starts whether it is set to 20, 60 or 145 secs

ricki-z, 2019-09-03:
Options (point 2) are added in beta branch.
Point 3 needs some more time.

MagTun, 2019-09-04:
Thanks so much for point 2! You are making us very happy!

dirkmueller, 2019-11-04:
The firmware with these options has been released. can this be closed?

MagTun, 2019-11-05:
Awesome! That's perfect.

closed 2019-11-05 by MagTun