opendata-stuttgart/sensors-software #446: Build fails with error "TypeError: a2b_base64() argument 1 must be string or buffer, not None"

nospam2000, 2019-09-02:
When building with PlatformIO, the after-build step fails with the following error message:
```
Building .pio/build/nodemcuv2/firmware.bin
Creating BIN file ".pio/build/nodemcuv2/firmware.bin" using ".pio/build/nodemcuv2/firmware.elf"
after_build([".pio/build/nodemcuv2/firmware.bin"], [".pio/build/nodemcuv2/firmware.elf"])
*** [.pio/build/nodemcuv2/firmware.bin] TypeError : a2b_base64() argument 1 must be string or buffer, not None
Traceback (most recent call last):
  File "/Users/michael/.platformio/packages/tool-scons/script/../engine/SCons/Action.py", line 1054, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "/Users/michael/Downloads/sensors-software/airrohr-firmware/platformio_script.py", line 17, in after_build
    target_name = b64decode(ARGUMENTS.get("LANG"))
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/base64.py", line 75, in b64decode
    return binascii.a2b_base64(s)
TypeError: a2b_base64() argument 1 must be string or buffer, not None

```

Before that error the following related warnings are shown in the console:
```
> Executing task: platformio run --environment nodemcuv2 <
Warning! Ignore unknown configuration option `lang` in section [env:nodemcuv2]
Warning! Ignore unknown configuration option `lang` in section [env:nodemcuv2_bg]
Warning! Ignore unknown configuration option `lang` in section [env:nodemcuv2_cz]
[...] // for all languages

```

See also the complete build log:
[build.log](https://github.com/opendata-stuttgart/sensors-software/files/3564932/build.log)

The script platformio_script.py uses `ARGUMENTS.get("LANG")` to get the command-line parameter "LANG", but "ARGUMENTS" doesn't contain "LANG". The content of ARGUMENTS is:
`{'PLATFORM_MANIFEST': 'L1VzZXJzL21pY2hhZWwvLnBsYXRmb3JtaW8vcGxhdGZvcm1zL2VzcHJlc3NpZjgyNjYvcGxhdGZvcm0uanNvbg==', 'ISATTY': '1', 'PROJECT_CONFIG': 'L1VzZXJzL21pY2hhZWwvRG93bmxvYWRzL3NlbnNvcnMtc29mdHdhcmUvYWlycm9oci1maXJtd2FyZS9wbGF0Zm9ybWlvLmluaQ==', 'PIOENV': 'bm9kZW1jdXYy', 'PIOVERBOSE': '0', 'BUILD_SCRIPT': 'L1VzZXJzL21pY2hhZWwvLnBsYXRmb3JtaW8vcGxhdGZvcm1zL2VzcHJlc3NpZjgyNjYvYnVpbGRlci9tYWluLnB5'}
`

The PlatformIO script `~/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py` doesn't set the "LANG" member, it only sets the following members:
```
clivars.AddVariables(
    ("PLATFORM_MANIFEST",),
    ("BUILD_SCRIPT",),
    ("PROJECT_CONFIG",),
    ("PIOENV",),
    ("PIOTEST_RUNNING_NAME",),
    ("UPLOAD_PORT",)
)

```

It is unclear to me where the member "LANG" should come from, that means which code or configuration option shall set it.

I downloaded the 'main' branch today and update PlatformIO and all libraries also today.

nospam2000, 2019-09-02:
I adapted the script according to the [PlatformIO documentation](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#custom-options-in-platformio-ini) and it works now. A pull-request will follow.

dirkmueller, 2019-09-02:
This happens when using platformio >= 4.0.0 - there was a pull request to limit to older versions for now recently

nospam2000, 2019-09-02:
1. Is the fix implemented in the correctly?
2. Do you see a better way to fix the problem?
3. Does the fix also work with platformio <4.0.0?
4. How long can you stay with platformio <4.0.0?

The fix itself actually only affects developers which use Visual Studio Code with PlatformIO, **it doesn't change the content of the built image**. The developers need the fix to do their work and get everything working with new versions (or ignore the error when you work with a single language).

nospam2000, 2019-09-02:
By the way: the next problem which will come up is that the linker script "eagle.flash.4m.ld" used in platformio.ini is incompatible in version 2.5.2 of esp8266 because there it has a SPIFFS size of zero, so the new script "eagle.flash.4m3m.ld" has to be used instead.

I'm still fighting to do a local build which is compatible with the address and size of the SPIFFS contained in the officially distributed images.

nospam2000, 2019-09-04:
fixed in beta branch

closed 2019-09-04 by nospam2000

modem-man-gmx, 2024-10-17:
regarding the plenty warnings
`Warning! Ignore unknown configuration option `lang` in section [env:nodemcuv2]
`
I learned the lang = ... statement is required every time you change language content. Disabling it, breaks language and/or UI maintenance.