opendata-stuttgart/sensors-software #470: Data writing to sd card?

jordanc365, 2019-09-12:
Has anyone had any luck writing data from the sensor directly to an SD card?
I'm planning on putting some sensors out in remote locations in regional Australia and it would be nice If I could still collect data if cellular went down for whatever reason.

Cheers
Jordan

Adorfer, 2019-09-12:
How dow keep track of timestamps? add a gns-receiver? (hint: there is not RTC in ESP8266, unless you migrate to ESP32)

executer-uno, 2019-09-15:
https://en.wikipedia.org/wiki/DCF77
That would be nice usercase for such clock-source. Much cheaper than gps. I just doubt about if such kind of signals available in Australia

jordanc365, 2019-09-16:
I was just going to use something like these with both an SD card and RTC on the same board. I just don't know how or where to code it up...

https://www.adafruit.com/product/1141
https://www.adafruit.com/product/2922
https://www.amazon.com.au/ACROBOTIC-ESP8266-Real-Time-Arduino-Raspberry/dp/B07NKGBDKH

Adorfer, 2019-09-17:
> Much cheaper than gps.

if you see devices with direct RS232/i2c which are cheaper than GPS-modules for 4$USD
Please adise!

[image]

nospam2000, 2019-09-17:
There is a possible solution without any external hardware clock.
When collecting data without having a valid time, just store the value of millis() (with counting overflows and only the full seconds) together with your measurement. Once you get an internet connection and get a time you can compare the difference between millis() and time and calculate the time for the stored values before sending them to the server.

Marwe, 2019-11-10:
Long long ago I experimented a bit with sdcards, but never brought it to work properly, so there is only a sketch and some pointers here (which might be outdated):
https://github.com/opendata-stuttgart/sensors-software/tree/master/esp8266-arduino/sensordev/sdcard