opendata-stuttgart/sensors-software #474: WDT not fires

executer-uno, 2019-09-15:
Very rare times board stops to response. I made small investigation on my own branch version of SW - it stucks somewhere inside SDK client function calls and no WDT fires occurs. I thought about external watchdog IC, but after found proper google request I found that solution:
https://github.com/esp8266/Arduino/issues/1532
It not tested by years, but looks fine and extra WDT should not be a problem. I just want to advice it to implement.
Thanks for your project!
PS: Also ESP8622 not supports WDT period customization, Im not sure about ESP32. But you trying to define it in the code, it can confuse someone)

nospam2000, 2019-09-21:
Can you please provide more information?

- Did you compile the firmware yourself?
  - If yes, did you use the Arduino IDE or VisualStudio Code?
- How is your configuration?
  - Which sensors, APIs and LCDs do you have enabled (the easiest way is to save the config and copy the resulting web-page as text)?
- Which language of the firmware do you use?

dirkmueller, 2019-11-27:
Please reopen when information.can be provided. Thanks!

closed 2019-11-27 by dirkmueller