opendata-stuttgart/sensors-software #499: mismatch in DNMS sendLuftdaten prefix removal

dirkmueller, 2019-09-25:
```
static void fetchSensorDNMS(String& s);
```

is creating sensor values in the form:

```
		add_Value2Json(s, F("noise_LAeq"), last_value_dnms_laeq);
		add_Value2Json(s, F("noise_LA_min"), last_value_dnms_la_min);
		add_Value2Json(s, F("noise_LA_max"), last_value_dnms_la_max);
```

however the sendLuftdaten call is filtering them as:

```
		if (cfg::dnms_read && (! dnms_init_failed)) {
			data += result_DNMS;
			sum_send_time += sendLuftdaten(result_DNMS, DNMS_API_PIN, FPSTR(SENSORS_DNMS), "DNMS_");
```

Which means either the "DNMS_" prefix should be "noise_" in the sendLuftdaten call, or the  add_Value2Json() should be calling the variables DNMS_noise_... instead.

which one is true?

ricki-z, 2019-09-26:
add_Value2Json() should be calling the variables DNMS_noise_... . The sendLuftdaten() call will remove the prefix "DNMS_". This prefix is a hint for other APIs. So they can tell apart the different sensors.

dirkmueller, 2019-09-30:
Has been fixed in beta

commit 0a2c101b (2019-09-26): Merge pull request #502 from dirkmueller/dnms_fix

Add missing prefix for DNMS values (Fixes #499)
commit 2b29e2d9 (2019-09-26): Add missing prefix for DNMS values (Fixes #499)

add_Value2Json() should be calling the variables DNMS_noise_... which
is a hint for other APIs so that they can tell apart the different sensors.
closed 2019-09-30 by dirkmueller