opendata-stuttgart/sensors-software #513: NRZ-2019-124-B7: Adafruit Si7021 / HTU21D stopped working after update to this version

zdroyer, 2019-10-09:
After the last beta update on October 1st, 2019, my temp+hum. sensor Si7021 (https://www.adafruit.com/product/3251 , HTU21D compatible),  stopped working.

I changed the sensor to a no-name clone and it worked (and still is working). 
But I tested the "faulty" sensor with another NodeMCU, and it seems that the 2019-124-B7 firmware is "faulty": when I switched back to the stable firmware (in the config), the Adafruit sensor works properly. I switched the firmware stable<->beta7 several times and the result is always the same: stable - works, beta - doesn't.

It seems the htu21d.begin method of the sensor fails (line 3937). 
Here is the serial debug output:
```Airrohr: NRZ-2019-124-B7
mounting FS...
opened config file...
parsed json...
output debug text to displays...
6
Connecting to ****
..............
WiFi connected
IP address: 192.168.43.46
Setting time using SNTP
Thu Jan  1 08:00:07 1970

..........
NTP time  ... not found
twoStageAutoUpdate
Success downloading: /airrohr/beta/latest_en.bin.md5
NewFW md5: 90b0c22407b1ff707a94fadd7e0206a3
Sketch md5    : 90b0c22407b1ff707a94fadd7e0206a3
No newer version available.
Starting Webserver... 192.168.43.46

ChipId: *******
Read HTU21D...
Check HTU21D wiring
Send to :
luftdaten.info
Madavi.de```

ricki-z, 2019-10-10:
Could you please check the latest beta version NRZ-2019-124-B8?
We are using the Adafruit HTP21D library and I have tested the latest beta with an HTU21D. The values are read. Both APIs (Luftdaten and Madavi) show the values.

zdroyer, 2019-10-10:
B8 checked - still not working.
In the meantime I found the solution: https://github.com/opendata-stuttgart/sensors-software/pull/514

zdroyer, 2019-10-10:
The Si7021 is compatible with HTU21D in reporting the measured values, but returns 0x3A if asked for read register ( https://github.com/adafruit/Adafruit_Si7021/blob/master/Adafruit_Si7021.cpp vs. https://github.com/adafruit/Adafruit_HTU21DF_Library/blob/master/Adafruit_HTU21DF.cpp)
The stable version of the airrohr firmware simply ignored the result of htu21d.begin(), while B7 remembers it and sets a flag, which prevents from reading the values from the sensor later in the program. In #514 I just added additional check in case Si7021 is installed. Here is the debug output of the patched B8:

```
NRZ-2019-124-B8.1
mounting FS...
opened config file...
parsed json...
output debug text to displays...
6
Connecting to *****
.......
WiFi connected
IP address: 192.168.0.52
Setting time using SNTP
Thu Jan  1 08:00:03 1970

..........
NTP time  ... not found
Starting Webserver... 192.168.0.52

ChipId: 2491543
Read HTU21D...
Read Si7021 replacement for HTU21D ...
Found Si7120 instead of HTU21D
Send to :
luftdaten.info
Madavi.de
----
```

BTW.
acTLS is deprecated now. I tried to move to BeerSSL - it compiled but crashed on the first attempt to connect to opensensemap.org :-(

zdroyer, 2019-10-14:
B9 fixes the problem. 
Made a PR for the Adafruit library to fix the problem, as @dirkmueller  suggested:
https://github.com/adafruit/Adafruit_HTU21DF_Library/pull/16

commit 40a679f2 (2019-10-13): Fix GPS and Si7021 regression (Fixes #513)
closed 2019-10-14 by zdroyer