opendata-stuttgart/sensors-software #518: Better support for 1 MB modules

smartynov, 2019-10-16:
I was trying to build an airsensor based on ESP-201 module recently. It is a esp8266 with 1 Mb of flash.

Pre-compiled firmware binaries fail to boot on such device, because it tries to initialize SPIFFS beyond the flash size. I was able to recompile and successfully flash firmware by changing board setting. However, current code relies on NodeMCU board in many ways, so I had to change minor things here and there in sources.

My suggestion is to adapt the code so it can be easily compiled for various esp8266 boards other than NodeMCU, in particular 1 Mb boards. I understand that OTA will be unavailable for 1 Mb devices, however manual flashing might be enough for some enthusiasts.

ricki-z, 2019-10-16:
Most changes you need to do can be done in ext_def.h . There are the defaults for all config settings and also the pin definitions i.e. for I2C or the connection of the PM sensors.
We need some SPIFFS space to save the configuration. Otherwise users need to change the wifi credentials in the source code and need to compile this.
Why we don't support those ESP modules: There may be changes in the API or bug fixes that need a firmware update. Would you like to do this manually for thousands of devices, that are placed outside?

smartynov, 2019-10-17:
I do understand that you rely on modern 4MB modules as a "mainstream", however there are lots of enthusiasts who may want to try older ESPs. I'm not asking you to keep building and providing binary firmware for generic platform. I'm just suggesting to make minor changes to the code in this repository to enable the project to compile for other ESP boards. As far as I discovered, there are very few changes needed to establish such a compatibility:

1. The code uses `D0`..`D10` constants defined in NodeMCU board headers, this can be redefined somewhere under `#ifdef ARDUINO_ESP8266_GENERIC`

2. Current beta uses `#include "WiFiClientSecureAxTLS.h"` – however last year esp8266 Arduino core deprecated axTLS and changed to BearSSL (see https://github.com/esp8266/Arduino/pull/5366). I found that changing this to `#include <WiFiClientSecure.h>` and changing `_client = new axTLS::WiFiClientSecure;` to `_client = new WiFiClientSecure;` seems to work (but it probably needs more testing).

Actually, I can prepare a PR with such changes if you agree to it.

P.S. Addressing possible API changes – I hope you will keep the old API as long as possible for compatibility. You're not assuming all the devices worldwide have OTA enabled, are you? And there are devices not using code from this repo (Raspberry Pi-based, for instance) that uses the same API.

ricki-z, 2019-10-19:
At first: please use the beta branch for changes. This is the latest code.
The master branch contains the code of the actual release version
Then you should look at the issues and (more important) the pull requests. There you will see what we are working on.

There should be a part with definitionsf for D0 .. D10 for the Arduino boards. If you remove the #ifdef or change it in a way that all non-NodeMCU will use this these definitions will be used.
And you need to change the pins for the sensors (not the API-pins)
And this should be enough to compile for other ESP8266-based boards.

The actual master is using the axTLS implementation as the BearSSL needs too much flash space. With that an OTA wont be able. And also your board with less or equal than 1MByte may be have bot enough flash space for the firmware.
So @dirkmueller did "some" code optimizations in the beta branch and now the image size is small enough that we can switch to version 2.5.2 of the Arduino core for ESP8266 and BearSSL.
And: without some optimizations the BearSSL code is much slower than axTLS. Many more connections would be open at the same time on our servers, increasing RAM and connection usage up to a value where the server might become unresponsive (there are actually more than 9500 sensor active at any time).

We will keep the "old" API, but we may add additional endpoints for data input.

smartynov, 2019-10-19:
Thanks @ricki-z. I have no problem amending the code to make it work for me, I was just thinking about other possible users and cleanliness of the project code. Glad to hear you're moving to latest Arduino core. I do not want to make your development process harder in in any way, so I think we should close this issue.

closed 2019-10-19 by smartynov

ricki-z, 2019-10-20:
We moved all module dependent definitions to ext_def.h. so an adoption for other boards should be really easy.