opendata-stuttgart/sensors-software #519: NRZ-2018-123B no expert configuration?

osmtools, 2019-10-17:
After flashing `NRZ-2018-123B` with `luftdaten-tool.linux-x64` to NodeMCU v3 there's no option on http://192.168.4.1/config for any API settings. Where is the expert configuration gone?

[image]

BTW: no matter what firmware is choosen it will flash NRZ-2018-123B ?!

osmtools, 2019-10-17:
Sorry for this, but was in correlation with #346 and my Fritz!Box 7490.
After serial monitoring the esp8266 i found out that it doesn't connect to the router. So in the router i've just delete an old entry with the same mac address. Now it connects correctly and i've found the configuration on the running webserver.

closed 2019-10-17 by osmtools

ricki-z, 2019-10-19:
Just a comment to this:
As were running out of memory in AP mode (config mode) with all options enabled, we needed to "disable" some of them. After connecting to a wifi you can access by opening the IP address of the sensor in your network in a browser and change all settings.
Thats why we publish precompiled binaries for the most commonly used combinations (SDS011 + DHT22, SDS011 + BME280) where the mentioned sensors are set as active.

osmtools, 2019-10-20:
@ricki-z in deed, after i've looked in the code. Is it mentioned in the FAQ or Wiki that there's a webserver after initial configuration? All the screenshots and answers i've found i the main and complete configuration in ap-mode.

ricki-z, 2019-10-20:
We are working on new instruction pages. With that we should be able to include such changes easier. The actual website is a wordpress site. After the last larger updates we have some problems with that installation.