opendata-stuttgart/sensors-software #520: Konfiguration: An eigene API senden fehlt

barrio, 2019-10-20:
Entgegen der Dokumentation hier: 
https://github.com/opendata-stuttgart/meta/wiki/Konfiguration-der-Sensoren#im-software-sourcecode
und hier:
https://github.com/opendata-stuttgart/meta/wiki/APIs#an-eigene-api-senden
wird nach flashen der latest.bin mit dem flashing tool o. g. Option nicht im Menu der Konfiguration angezeigt, bei beta & de Versionen ebenso.

Wäre schön, wenn Ihr das wieder verfügbar machen könntet. Das flashen aus der Arduino IDE finde ich etwas aufwendig, nur um diese Konfiguration machen zu können.

ricki-z, 2019-10-20:
Hallo,
bitte verbinde den Sensor mit einem WLAN. Dann kannst du in dem entsprechenden Netzwerk über die an den Sensor vergebene IP-Adresse auf den Sensor zugreifen. Und dann ist der Konfigurationspunkt auch wieder sichtbar.
Es gab ein kleines Speicherproblem im Konfigurationsmodus, so daß wir die Konfig-Seite deutlich kürzen mussten.
Wir überarbeiten gerade die Seiten, in der nächsten Version sollte dies dann entsprechend erklärt werden.

barrio, 2019-10-20:
Danke für die schnelle Antwort, funktioniert!

dirkmueller, 2019-10-28:
@ricki-z so the reason for the shorter number of options in the AP mode was just memory constraints? well, that would make things simpler..

ricki-z, 2019-10-29:
@dirkmueller yes, we couldn't send / build the whole page in AP mode. And as a "bonus" most users didn't need to scroll through options they don't need to change. There were some users that changed options (activating/deactivating APIs and sensors) and were wondering why the device didn't send anything.

ricki-z, 2019-11-06:
Maybe we should add a line on the config page:
"More options can be accessed after the device is connected to your network."

dirkmueller, 2019-11-26:
More (most) settings should be available again in first time wizard with the next stable update.

commit 67fd482f (2019-11-11): Show more items in first config mode (Fixes #520)

Add most of the fields back for initial configuration mode.
closed 2019-11-27 by dirkmueller