opendata-stuttgart/sensors-software #521: BME 280 Temperature Calibration

22MarioZ, 2019-10-23:
Is possible to add to next revision the option to calibrate the temperature for the BME 280 sensor. I noticed that my sensor have 4-5 degree C offset compare to real temperature.

bertrik, 2019-10-23:
Looking for this issue on the internet, I find several threads / forum posts describing the same problem. Apparently under some circumstances, the BME280 can self-heat. I think it is possible to mitigate the self-heating in software. Would be worthwhile to investigate what the current software is doing w.r.t. avoiding self-heating.

ricki-z, 2019-10-23:
You don't need to investigate. We use a mode without self-heating. We had this problem in a very early stage of supporting the BME280 in our firmware.
The BME280 (like all other temperature sensors) should be placed as near as possible at a case opening. So the sensor can measure the outside air and not the waste heat.

dirkmueller, 2019-10-27:
We're using the factory calibration data for temperature compensation. Is this a new problem that appeared in a specific firmware release?

22MarioZ, 2019-10-28:
I'm using the latest release of the firmware. Trying to isolate the sensor from heat influence of ESP8266 or SDS011, It went down, but still 3.5 degree difference.

dirkmueller, 2019-10-28:
Yes, question was aiming at whether this sensor used to work with an older firmware version. Also, latest stable or latest beta version? Please be specific, there were significant changes.

22MarioZ, 2019-10-28:
Runing the NRZ-2018-123B firmware.

stewecar, 2019-10-28:
A trick could be to double the distance between the sensor and the nodemcu..

dirkmueller, 2019-10-28:
@22MarioZ please check the 'use beta version' checkbox and save the config (which retarts the sensor and updates it. Please be patient, this can take a few minutes). Let me know if that changes anything. If it doesn't I'll build a custom firmware dumping the calibration data.

22MarioZ, 2019-10-29:
I updated it with the latest version of the FW (NRZ-2019-124-B9). Still around 3 degree C difference. I keep everything in the box and the sensor have its own access to external air. It is isolated from rest of the box.. For one hour I keep sensor out of the box and I noticed the same behavior. Maybe for next revision of the FW you can add the temp offset parameter (like ESPEasy).

dirkmueller, 2019-10-29:
Thanks for testing, that's unfortunate to hear. I would like you to test a debug firmware that prints out the factory calibration data.

While it is possible to add a temperature offset compensation, it is important to realize that the other parameters measured by bme280 (humidity and pressure) are strongly temperature dependent, and their readings are depending on accurate temperature sensing. In fact temperature is not the main objective of the sensor, it is pressure.

As such changing the temperature reading will strongly modify the other measurements.

Any opinions whether the other readings of that sensor are plausible?

It is possible that the sensor is defective or that the factory calibration has been performed poorly for some reason. 

I did notice that you can sometimes.recover the sensor by dry-baking it in an oven (there are various descriptions for that, ~ 120°C for 1 hour at least). In the end a new sample might be the cheapest and quickest way to success though.

22MarioZ, 2019-10-29:
Let me know how to test the debug firmware first.. Lets see its results and then we go to next step.

bertrik, 2019-10-29:
I would say that the main purpose of the BME280 is to measure relative humidity. 
Getting temperature and pressure is just an extra.

When the software already does its best to get the best possible reading (using calibration data etc), then I think you should not try to "improve" it because you think it is wrong in some way. Temperature measurement is not super-trivial, for example you need to make really sure that you're really measuring the air temperature and have all kinds of radiative heat sources eliminated (for example). Adding extra configuration items (offset adjustment for example) just makes things more complicated, IMO. How sure are you that the other ways of reading temperature are more accurate anyway?

dirkmueller, 2019-10-29:
@22MarioZ when you flash the firmware from http://air.dmllr.de/airrohr/debug/latest_de.bin you should get output similar to this

```
Trying BMP280/BME280 sensor on 76 ... found
dig_T1: 28376
dig_T2: 26465
dig_T3: 50
```

I'd be interested in the exact values for dig_* in your sensor. perhaps that indicates some kind of problem in the code that we can fix.

22MarioZ, 2019-10-31:
@dirkmueller I tried the firmware, but the ESP8266 went into reboot loop when connecting to AP. The Captive portal pop-up, but newer displayed in the way I could enter WiFi credentials.

I also connected another BMP 280 sensor and DHT22 sensor and I have the same temperature from both sensors. I'm thinking to replace the current sensor with DHT22 or BMP280.

Thank you for all your help.

dirkmueller, 2019-10-31:
@22MarioZ thanks for the feedback, this is even more important. I will test this now first..

dirkmueller, 2019-10-31:
@22MarioZ I am not able to reproduce, but I built a new fw based on "stable" tree with just this modification. it is 

```
471280 31. Okt 19:27 latest_de.bin

$ md5sum latest_de.bin
24cedd0d28bd3fc87164d3e1febfd725  latest_de.bin
```

this one boots and connects and prints the calibration parameters.

dirkmueller, 2019-10-31:
@22MarioZ I was able to find and fix the problem with the crashloop!

22MarioZ, 2019-11-04:
@dirkmueller Didn't have time before and today uploaded the latest beta but I don't get the BME calibration values.  I noticed another issue. No data is send to openSenseMap. I don't if anybody else noticed this issue.

dirkmueller, 2019-11-04:
@22MarioZ the calibration values are only printed with the debug firmware mentioned earlier. I replaced it in place with a working version.

With the official beta for me opensensemap is working.

sanitariu, 2019-12-08:
My BMP280 also shows wrong values. Humidity - 33.6 %  is wrong and temp too. How i can recalibrate my sensor or see results from it via the web ?

dirkmueller, 2019-12-08:
Are you saying it shows negative humidity?

sanitariu, 2019-12-09:
Hum: 34.2 %
Temp: 17.1 °C
Pressure: 994.09 hPa
It is not calibrated. Current temperature is 4%, Humidity is over 50%.....
Is there anyway i recalibrate it ?

relyd, 2019-12-16:
As I understand the board design plays an important role in the self heating, especially if there is a LDO on the same board. There are numerous PCB designs, some better than others that may influence readings. Also handling of the sensor during transport and usage and interactions with chemical compounds that we cannot assess from the manufacturer may contribute to skewed readings. I would favor an offset in UI as well, because using a BME280 with 2degrees higher readings is not good for any public acceptance of data, even when rel. humidity and pressure might get influenced.  Reactions I hear are like : that thing doesn't even read the temperature right, so the rest must be off as well. 

@22MarioZ  Could you share an image of the specific sensor board in use?

dirkmueller, 2019-12-17:
I might consider adding a compensation, I just don't know whether it needs to be a fixed offset or some kind of linear factor.

Also humidity and pressure are indeed influenced by the temperature reading, temperature is directly used to calculate the compensated values.

dirkmueller, 2019-12-17:
I did notice that this sensor seems to be extremely easy to damage with too much heat during soldering, and then it reports values that are completely off

22MarioZ, 2019-12-17:
@relyd; I don't have any picture, but looks like this:
https://www.aliexpress.com/item/32833941464.html?spm=a2g0s.9042311.0.0.27424c4dblBMHT 

PS: I plan to replace it with HTU21D to see if I get more correct measurements...

dirkmueller, 2019-12-18:
Just fyi the recent firmwares have support for sensirion SHT 3x sensors, which are pretty good precision and reliability wise

dirkmueller, 2020-01-15:
I also recently added an option to adjust / compensate the temperature reading from sensors in the beta firmware version.

Is there anything else that can be done on this issue or are you good with closing it?

sanitariu, 2020-01-16:
I will test beta too. You can close

22MarioZ, 2020-01-19:
temp compensation is working, the ticket can be closed.

closed 2020-01-19 by dirkmueller