opendata-stuttgart/sensors-software #522: Library: SoftwareSerial (1.0.0) (GNU Lesser Public License> = 2.1

Uwana, 2019-10-25:
Hello,

Title refers please.

The recommended 1.0.0 version of the library above is no longer available on Arduino IDE.

Can the source code be revised to work with a more recent version of the library?

Thanks

ricki-z, 2019-10-25:
Hello,
we are using the SoftwareSerial that is shipped with the Arduino Core for ESP8266. The version of this core for the master branch is 2.4.2. In the beta branch we are preparing the switch to core 2.5.2. This core is blowing up the firmware image size by 50kByte, which would make OTA updates impossible. So we had to change the update process. But this had to be done with the "old" core.
This OTA update should be available in the next days, last tests are running. Then we will switch to the 2.5.2 core in the beta branch.
(At the moment this firmware is running on more than 11.000 devices. So we have to be very careful with updates)

Uwana, 2019-10-25:
Thanks for the feedback.

How would you advice folks using the arduino IDE to install the firmware to go about this, since the IDE's library currently shows only up to version 5.0.0

ricki-z, 2019-10-27:
You don't need to install a SoftwareSerial lib via the Arduino IDE. As mentioned the lib included in the Arduino Core is used. In the README.md of the actual beta version you will find an appropriate hint. The "problems" with newer versions of the SoftwareSerial started after the upgrade to 2.5 of the Arduino Core for ESP8266.

dirkmueller, 2019-10-27:
The issues with newer SoftwareSerial have been solved in 606781601f11b295cf52df81f48a216371d7c7a0

dirkmueller, 2019-10-27:
@Uwana The issue you're hitting has been fixed already in the beta/ branch (the Readme has been corrected). Hopefully there is a release to stable (merge to master/ branch) soon.

dirkmueller, 2019-11-01:
Has been released in yesterday's stable update

DeeKey, 2019-11-03:
@ricki-z can this issue be closed?

ricki-z, 2019-11-03:
@DeeKey you should ask @Uwana . He opened the issue ;-)

closed 2019-11-03 by Uwana