opendata-stuttgart/sensors-software #525: Prometheus metrics api broken with NRZ-2019-125

mrMoe, 2019-10-29:
Hi there,

last sunday my sensor automatically updated to version NRZ-2019-125 despite having stable releases selected.
Since then I'm not able to get PM2.5 or PM10 measurements and the prometheus metrics api is broken. The id string is generated incorrectly so prometheus will not parse the api.

software_version{version="NRZ-2019-125",node="-12345678"} 1
uptime_ms{12345678} 569382
sending_intervall_ms{12345678} 240000
number_of_measurements{12345678} 2

Since version 125 seems not to be commited I could not verify any changes in the source code.

Why did my sensor update to a version which is not commited?

Thanks for any help!!

ricki-z, 2019-10-30:
It seems that none of our 300 sensors, that use the beta version, is access this prometheus endpoint.
The version NRZ-2019-125 should be the next release version and is identical with the actually latest beta branch version. There I have changed the endpoint so that it should work again. If you restart your sensor it should update to that version.

mrMoe, 2019-10-30:
Hi,
I just restarted the sensor and the version is stil on Firmware: NRZ-2019-125.
The reading of the PM2.5 seems to work now but the metrics are still broken.

```
software_version{version="NRZ-2019-125",node="esp8266--12345678"} 1
uptime_ms{12345678} 258112
sending_intervall_ms{12345678} 240000
number_of_measurements{12345678} 1
SDS_P1{node="esp8266-12345678"} 13.60
SDS_P2{node="esp8266-12345678"} 4.50
temperature{node="esp8266-12345678"} 21.30
humidity{node="esp8266-12345678"} 60.00
samples{node="esp8266-12345678"} 3802099
min_micro{node="esp8266-12345678"} 54
max_micro{node="esp8266-12345678"} 85181
signal{node="esp8266-12345678"} -25
last_sample_age_ms{node="esp8266-12345678"} 15882
```
The node identification string in the first 4 lines is incorrect which hinders prometheus to parse the metrics

ricki-z, 2019-10-30:
I'm sorry that there is no device using beta versions and used with prometheus. So we haven't seen this earlier.

PR #526 (2019-10-31): fix: remove duplicate node label. fixes #525
Fixes #525 broken prometheus api by removing duplicat node label from software metric
commit d4fe275b (2019-10-31): fix: remove duplicate node label. fixes #525
commit f1278d95 (2019-10-31): Merge pull request #526 from mrMoe/prometheus-metrics-#525

fix: remove duplicate node label. fixes #525
closed 2019-10-31 by ricki-z