opendata-stuttgart/sensors-software #536: Unable to downgrade from Beta to stable in menu.

DeeKey, 2019-11-03:
Remove check for "Load Beta software"
Choose any language  (this will initiate firmware download).
Expected result - new stable version will be flashed.

Actual:
unit rebooted, but version remains the same with the same language as before.

dirkmueller, 2019-11-03:
I can't reproduce, for me this works (did a full stable->beta->stable cycle with and without language change).

Please note that you don't need to change the language in order to trigger a change if firmware. It works without language change.

Is there a firewall between the logger and the internet? Only reason I can think of is that it fails to contact the update server.

If you can collect the serial debug output please attach it here.

bertrik, 2019-11-04:
This is what my sensor says on serial:
(Firmware version: NRZ-2019-126-B2)

ws: config page ...
output debug text to displays...
Saving config...
Before writing config..
Config written successfully.
Restart.
H!�1G�
��Airrohr: NRZ-2019-126-B2
mounting FS...
opened config file...
parsed json...
output debug text to displays...
6
Connecting to Freifunk-disabled
......
WiFi connected
IP address: 192.168.0.110
Setting time using SNTP
Thu Jan  1 08:00:03 1970

....................
NTP time  ... not found
twoStageAutoUpdate
Firmware download failed!
Starting Webserver... 192.168.0.110

ChipId: 9137604
Read PMS(1,3,5,6,7)003...
Stopping PMS...
Read BMP280/BME280...
Trying BMP280/BME280 sensor on 76 ... found
Send to :
luftdaten.info
----
Auto-Update active...

dirkmueller, 2019-11-04:
> Connecting to Freifunk-disabled

There's a WiFi with that name??!

So the issue is that there is no NTP, so the node starts with system time of 1970, where it fails to connect to the update server because the server tls certificate is not valid.

I  guess we'll have to find a workaround for that.

dirkmueller, 2019-11-04:
We could try to set the system time to something slightly newer than the compilation timestamp of the currently running firmware.

Another way would be that we switch off tls if we have a broken system time.

bertrik, 2019-11-04:
Ah, having no time explains a lot. Indeed, out of convenience I set up a hidden network with that name on my wireless router, it has full access to the internet (no firewall/filter, etc, just NATed). At the very least, I think we could log the time server name we are connecting to for NTP.

dirkmueller, 2019-11-04:
@DeeKey it uses hardcoded two servers: a) the local wifi gateway IP and b) the $langcode.pool.ntdp.org server, where $langcode refers to the firmware's (not configured!) language. 

while the first one might not work, the 2nd one should always work if there is udp access to the internet possible. Potentially in your case the 1st one does not answer and we're not waiting long enough to query the 2nd one or something like that. 

I'll try to simulate this in a testenv.

dirkmueller, 2019-11-05:
@DeeKey ok, good that we found this edgecase, I actually spotted in a testenv multiple bugs with this. once above mentioned PR is merged, you have to patch to the newer beta firmware once manually by e.g. downloading it from https://firmware.sensor.community/airrohr/beta/latest_XX.bin and then this issue should be solved.

dirkmueller, 2019-11-08:
@DeeKey can you confirm that things are working for you regarding downgrading when you start from either stable or current beta?

dirkmueller, 2019-11-26:
@DeeKey any response?

dirkmueller, 2019-11-27:
Please reopen if there is still an issue.

PR #539 (2019-11-04): Do not validate firmware update server certificate when system time i…
…s wrong (Fixes #536)

Before we downloaded the firmware unencrypted.. it seems time
is difficult to obtain. Lets not lock ourselves out.

Also print the NTP servers that are being used.
commit 9e9c95c1 (2019-11-04): Do not validate firmware update server certificate when system time is wrong (Fixes #536)

Before we downloaded the firmware unencrypted.. it seems time
is difficult to obtain. Lets not lock ourselves out.

Also use proper waiting time for NTP and use only generic servers,
as for some the gateway does not respond with NTP, and there are
language variants that don't have DNS (e.g. en.pool.ntp.org).
closed 2019-11-27 by dirkmueller