opendata-stuttgart/sensors-software #537: Error with data transmission to Feinstaub-App server

mrgames13, 2019-11-03:
In the latest beta version there is a bug with the data transmission to the Feinstaub-App server (h2801469.stratoserver.net). Were there changings, which could lead to that error?

I hope we can manage this issue until the version goes stable and will be distributed to all users. We already receive one email every minute.

Thanks in advance!

dirkmueller, 2019-11-03:
It is possible that there were changes.. which error do you have?

mrgames13, 2019-11-03:
All we know is, that version NRZ-2019-125-B1 worked well and since then, our server does not receive data any more. From no sensor that has been updated to one of the next versions. So we assume, that there changed something, when sending the data to our server.

dirkmueller, 2019-11-03:
I think I found something..

dirkmueller, 2019-11-03:
Should be fixed by the PR above.

mrgames13, 2019-11-03:
Great! Thanks a lot. Only the release of the update is missing ;)

ricki-z, 2019-11-03:
If I read the changes, it should also work if there were a 2048 bit certificate for https at the FsApp server (larger certs will cause a memory overflow).

mrgames13, 2019-11-03:
That's right. Should also work on port 443. Maybe the port change does not fix the problem described above?!

ricki-z, 2019-11-04:
I have checked the master branch source code. There is a forced unencrypted connection used for the FsApp.
I will push a new beta (will have the same number, but should update on sensor restart).
Could you please test this with that version?

mrgames13, 2019-11-04:
No, that does not work.

ricki-z, 2019-11-04:
Did you switch to beta?

mrgames13, 2019-11-04:
Sure

ricki-z, 2019-11-04:
The transmission to the FsApp server is using the same format we use to transmit data to Madavi.de. And there it's working ...
Can you see any error messages on your server?

mrgames13, 2019-11-04:
I checked the server logs. There is no single request for today, which does not have the result code 200. It seems so, if the requests would not reach the server.
I recognized a strange thing. It seems, that one single sensor located in Malaysia / George Town is able to send its data with firmware NRZ-2019-126-B3. The server log contains several occurences of this line:

[04/Nov/2019:00:29:18 +0000] "POST /data.php HTTP/1.1" 200 261 "-" "NRZ-2019-126-B3/393834"

mrgames13, 2019-11-04:
I've tested around a bit and I tried to enter the server host, path and port into the corresponding fields for the custom API endpoint. Following configurations worked:

Host: h2801469.stratoserver.net (with or without www. at the beginning)
Path: /data.php
Port: 443 (HTTPS option above enabled)

And

Host: h2801469.stratoserver.net (with or without www. at the beginning)
Path: /data.php
Port: 80 (HTTPS disabled)

dirkmueller, 2019-11-04:
it does work for my two sensors with the patch. the issue before was that it sent *unencrypted* http to port 443, which just produces garbage and is being rejected. Now it sends to port 80 unencrypted http, which is the same state like the stable firmware and it seems to work. I can see data submissions of my beta sensor in the app that are current.

mrgames13, 2019-11-04:
Okay. I tested it again and it works now. Thank you for your help!
@ricki-z maybe you can include the https support as we have agreed per mail.

mrgames13, 2019-11-08:
Hi again,
I have to reopen the issue, cause the bug appeared again. Since version NRZ-126-B4 the data gets lost on its way again. As I mentioned, if you configure the optional API like that:

> Host: h2801469.stratoserver.net (with or without www. at the beginning)
> Path: /data.php
> Port: 443 (HTTPS option above enabled)
> 
> Or
> 
> Host: h2801469.stratoserver.net (with or without www. at the beginning)
> Path: /data.php
> Port: 80 (HTTPS disabled)

it works like a charm.

Looking forward to your answer!

dirkmueller, 2019-11-08:
Confirmed. Weird. I thought I properly tested this.

What changed is that I enabled tls. And likely the server does not support 'Maximum Fragment Length Negotiation' which we require because it reduces memory pressure and works for all the other predefined APIs

dirkmueller, 2019-11-08:
actually using openssl s_client -connect h2801469.stratoserver.net:443  gives

```
    Start Time: 1573226278
    Timeout   : 7200 (sec)
    Verify return code: 10 (certificate has expired)
```

for me. we have TLS verification turned on. I verified that turning off verification makes it succeed sending data again.

mrgames13, 2019-11-09:
Thanks for the fix. @ricki-z Could we merge this into beta?

ricki-z, 2019-11-10:
Is merged in beta and uploaded to the update server.

mrgames13, 2019-11-10:
Thanks! I tested it and it seems to work? It takes 24 hours until all beta users have this update, right?

ricki-z, 2019-11-10:
Yes, the sensors should update in 24 hours. But they will also update after a restart.

mrgames13, 2019-11-10:
Awesome. Thanks for the fix!

commit 9e1b0371 (2019-11-03): Merge pull request #538 from dirkmueller/beta

Fix copy&paste typo on FsAPP submission (Fixes #537)
commit e24e9f2f (2019-11-03): Fix copy&paste typo on FsAPP submission (Fixes #537)
commit 12c37063 (2019-11-10): Disable TLS verification for Feinstaub App (Fixes #537)

Seems like the server certificate is failing validation recently.
commit 7e995b7c (2019-11-10): Merge pull request #543 from dirkmueller/beta

Disable TLS verification for Feinstaub App (Fixes #537)
closed 2019-11-10 by mrgames13