opendata-stuttgart/sensors-software #541: WiFi signal spiking to +30 dBm

RikDrabs, 2019-11-05:
When reading back the graphics of the WiFi signal value & quality, WiFi signal spikes up to +30 dBm appear at random times. With the same timing measurement values disappear. 
The sensors are: 
https://www.madavi.de/sensor/signal.php?sensor=esp8266-11730516
https://www.madavi.de/sensor/signal.php?sensor=esp8266-3134159
https://www.madavi.de/sensor/signal.php?sensor=esp8266-9233732  (now out of service)
https://www.madavi.de/sensor/signal.php?sensor=esp8266-3951554
Any idea what is happening ? Weather was a lot colder recently, in sync with the problems ?
Software ? Hardware ? Temperature ? Or a combination of these ?

dirkmueller, 2019-11-06:
so the "spike to 100%" is a software issue in our firmware I think, and it actually means "we lost the connection". I think the function we're calling is returning "0" when not connected, which is interpreted (falsely) as 100% signal strength.

 the more interesting value is the dbm, and it shows in those graphs that is consistently very low. it needs to be higher than -80dbm (so in the range of -80dbm - -30dbm) for stable connectivity. in your graphs you're in the -90dbm and -80dbm range, so that is not good enough.

what we can do is tell the wifi phy to use a lower connection speed on low connectivity quality in the hope that it helps, but fundamentally you need to work on antenna positioning / distance to the wifi access point.

RikDrabs, 2019-11-09:
The dBm value is indeed low, around -90 dBm, in all sensors producing this error. One more sensor appeared today with the error. What i don't understand is that the WiFi signal is apparently strong enough to get these +30 dBm values transmitted, while at the same time the program in the MCU decides that the signal value is too low and replaces that low value with +30 dBm. There is something contradictory ...

RikDrabs, 2019-11-09:
Closed by mistake.

dirkmueller, 2019-11-09:
according to 3.5.29. in the sdk reference manual, 31 is an "invalid value measured" error reporting. so it is a special error code and does not actually mean +30 dBm .

RikDrabs, 2019-11-11:
But how can the signal value be measured in an invalid way? This condition starts occuring when outside temperatures drop to around freezing. With this condition, the program stops transmitting measured PM's, but still transmits the +30 dBm value, which is the error code 31 according to your explanation; so the connection is still there, even at signal values below -90 dBm. This is not normal behaviour and instead looks more like a (temperature related ?) bug. Or is this already corrected in the beta branch ?

ricki-z, 2019-11-13:
I have changed the graphs at madavi.de so that 31dBm is handled as "unknown value".
The graphs should show the real "conditions" much better now.

RikDrabs, 2019-11-14:
Can this be closed and superseded with issue #384

closed 2019-11-14 by RikDrabs