opendata-stuttgart/sensors-software #544: BME 280 liefert nur noch 100% Luftfeuchte

dora71, 2019-11-09:
Seit dem Firmware Update auf Version NRZ-2019-125-B1 liefert mein BME 280 nur noch 100% Luftfeuchte.

Temperatur und Luftdruck sind ok.
Ein Upgrade auf die Beta Software NRZ-2019-126-B5 brachte keine Verbesserung.

An der Verkabelung wurde nichts geändert.

BudBundi, 2019-11-09:
Ich dachte es auch lange bei mir, aber ich habe auch einige Werte über 90% dabei. Allgemein scheint dort vielleicht irgendeine Umrechnung nicht zu passen.

dirkmueller, 2019-11-09:
[image]

dirkmueller, 2019-11-09:
so in general a value of near 100% is not unusual at this time of the year.

sanitariu, 2019-11-25:
I have the same. I think it is a bug in the firmware. Developers must return some version back so all stations download/update automatically. Sensors SDS011 and BME 280 seems do not work well with the latest firmware.

dirkmueller, 2019-11-26:
@sanitariu are you having the issue described here which is 100% humidity always or something else?

sanitariu, 2019-11-26:
I have humidity now 31.4 % and it is raining. Also temperature is +2 degree but sensors shows 23.2 °C.
I would like a way to return to older version like some from 2018 without going to the roof and dismount everything. How i can do it ? Also other sensor for dust does not show correct values. This started from some days after latest update.
But yes it do not show 100%

dirkmueller, 2019-11-26:
@sanitariu ok, so it is a different issue. please create a new one, it is free :-) I'm trying to help you there. Regarding the question, there is no builtin downgrade support for OTA. we assume that newer is better. please try the BETA version (check both the "use update" as well as "use beta" flag in configuration and save). I believe the issue is fixed already.

dirkmueller, 2019-11-27:
I can not reproduce this with my set of BME280 sensors. Any user with *this* issue please share details.

ricki-z, 2019-11-27:
And if possible please send me a picture of the module to rajko (at) sensor.community.

dirkmueller, 2019-12-30:
please reopen when information can be provided.

closed 2019-12-30 by dirkmueller