opendata-stuttgart/sensors-software #551: Neues Firmware-Update löst Probleme aus & Maps/Karten seit Tagen nicht erreichbar

iceweasel1, 2019-11-14:
Abend allerseits,

kann es sein das die vergangenen Tage automatisiert ein Update der Firmware aus- bzw. eingespielt wurde? Vor ein paar Tagen sah die „values“-Übersicht noch etwas anders aus in meiner Erinnerung. Jetzt steht am Ende unter „Anzahl Messungen:“ als neue Zeile „Dauer Messübertragung:“ dabei. Im mittleren Bereich steht nicht mehr (nur) „BME280“, sondern „BMP/E280“. 

_Frage: Sind die aufgezählten Dinge Neuerungen in dem (möglichen) Update der Firmware?_

Seitdem es mir die Tage die neure Übersicht anzeigt, spuckt es mir bei den SDS011-Werten Dinge aus, die bislang nie so zu sehen waren. Entweder die Werte bei 2.5 & 10 zeigt es schlichtweg überhaupt nicht mehr an (einfach Bindestriche), oder Werte, die mehrere hundert µg/m³ betragen. Zumindest seit April, seitdem ich den Sensor laufen habe, gab es nie gravierende Schwankungen, bzw. kein einziges Mal „Bindestriche“, also keine Werte zu sehen.

_Frage: Ist dies absichtlich so/rührt das von dem (möglichen) neuen Update her, die eventuell einen unbeabsichtigten Bug enthält?_

**Offtopic**

Überschrift bei „https://luftdaten.info/laermsensor-bauen/“ steht noch „Feinstaubsensor“. Ich vermute, dass es Lärmsensor, wie es im restlichen Text etc. geht, heißen soll.

Gibt es derzeit Probleme mit der Karte? Egal maps… oder deutschland oder ein anderes Land davor, keine der Links geht seit Tagen mehr. Egal mit welchem Browser, es kommt sofort eine Fehlermeldung.

Grüße
René

**Anhänge**
---
[image]
[image]
[image]
[image]
[image]
[image]

dirkmueller, 2019-11-14:
@iceweasel1 ja, seit oktober 31st ist ein firmware update ueber OTA installiert worden, dadurch gab es neue features (u.a. SPS30 und Laermsensor support) und viele andere Verbesserungen. 

Details siehe Changelog https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/Versions.md

dirkmueller, 2019-11-15:
Duplicate von #552 . Kann dieses issue geschlossen werden?

iceweasel1, 2019-11-15:
@dirkmueller Eigentlich kein Duplicate. Das war knapp 2 Wochen bevor ich #552 die Tage eröffnet hatte.

@ricki-z hatte gestern dieses und das andere Theard in /sensors-software verschoben. Hatte das versehentlich in /meta eingetragen.

ricki-z, 2019-11-15:
Zum Problem mit maps.luftdaten.info siehe Antwort unter #255 .

iceweasel1, 2019-11-15:
@ricki-z stehe gerade etwas auf dem Schlauch was du mir damit sagen möchtest (sorry). Habe #255 plus die Verlinkung von dort zu #256 eben durchgelesen. 

Das mit dem Hinweis zu der (Maps-)Domain bestand nicht nur da wo ich es hier am 01.11.2019 um kurz vor 21 Uhr geschrieben hatte, das war mehrere Tage schon so gewesen, das es nicht mehr zu erreichen war und ich es daher dann mit der Firmware-Sache als Offtopic nebenbei erwähnen/hinweisen wollte. ;-)

dirkmueller, 2019-11-15:
ich denke das war ein tippfehler, @ricki-z meinte #552

ricki-z, 2019-11-15:
Ja, da hatte ich mich vertippt. Sollte #552 sein. Sorry, aber ich wollte zumindest den Teil "lösen".

dirkmueller, 2019-11-27:
diese issues und #552 sind zu dem gleichen thema, bitte die diskussion in #552 weiterfuehren!

closed 2019-11-27 by dirkmueller