opendata-stuttgart/sensors-software #552: Ausgerolltes Firmware-Update führt zu massiven Problemen mit Sensor & Messung

iceweasel1, 2019-11-14:
Hallo nochmals,

ich hatte bereits vor 12 Tage einen Hinweis und Fragen [hier ](https://github.com/opendata-stuttgart/meta/issues/122) gestellt. Die Problematik ist mehr geworden als mir lieb ist. Aus den Socken haut es mich, als ich eben die E-Mail (vermute automatisiert versendet) erhalten habe, das mein Sensor angeblich seit mehr als 24 Stunden keine Daten mehr in die Datenbank gesendet hätte. Die sofortige Überprüfung ergab, das es eine 50/50 Sache ist. 

Seit gestern (12.11.2019) nach 07.03 Uhr sind bis jetzt keine PM-Daten mehr zu sehen. Durch Grafana kann ich es auf diese Uhrzeit eingrenzen. Was jedoch keinen 24h-Ausfall anzeigt, ist das WLAN-Signal und die BME280, also Temperatur, Luftfeuchte und Luftdruck. Diese Daten sind bis jetzt soweit im Archiv und Grafana zu sehen.

Was ich im anderen Thread vor 12 Tagen beschrieben hatte, war, das seit ein neues Firmware-Update (automatisiert) ausgespielt wurde, es meinem gebauten Sensor letzten Endes und mittlerweile massive Probleme bereitet. Die Sache, die ich erwähnt hatte, das die PM2.5 & 10-Angabe in der Übersicht, wenn man diesen im Netzwerk über die IP-Adresse ansteuert, leider oft leer bleiben/keine Zahlen mehr, sondern Bindestriche anzeigen. Das hat sich die vergangenen 12 Tage sehr oft wiederholt, bis jetzt, das es gar nichts mehr anzeigt. 

Da die E-Mail mit dem Hinweis der "... mehr als 24 Stunde..." jetzt auch noch kam, bestätigt sich der Verdacht und ebenso die Beobachtung der letzten 12 Tage, dass, sobald es einfach diese Bindestriche anzeigt, eine Übertragung in die Datenbank zu 50 % stattfindet und ebenso 50 % nicht stattfindet. Dies konnte ich oft beobachten, das mal bei angezeigten Bindestrichen die Daten dennoch eingespielt wurden, und leider wieder nicht. Im Archiv dann von der notierten Uhrzeit bei der Nachschau zu überprüfen gewesen. 

Das Thema, dass ebenfalls angesprochen wurde, das bei "Anzahl Messungen" es nach mehreren hundert, wieder und wieder mal bei 0 anfängt, obwohl kein manueller oder automatisierter Neustart ausgeführt wurde, besteht weiterhin. 

Ich führe die gesamten genannten aufgetretenen Probleme auf das ausgerollte Firmware-Update raus. Warum auch immer, aber anscheinend verträgt es mein nach der Anleitung gebauter Sensor nicht. Seit meinem Beginn April bis zum vor kurzem Update, gab es kein einziges Problem bzw. Probleme mit der Firmware. 

Das war das vorherige Thread: https://github.com/opendata-stuttgart/meta/issues/122

Kommt da ein gefixtes Update?

[image]
[image]

dirkmueller, 2019-11-14:
@iceweasel1 um die beta zu installieren, muss der haken bei "auto update" *und* bei "benutze beta" gesetzt werden (der erste ist der on/off switch fuer den zweiten). 

Ich gebe dir recht dass die ui das sichtbarer machen koennte. aber das ist erst mal zweitrangig.

dirkmueller, 2019-11-14:
In https://www.madavi.de/sensor/signal.php?sensor=esp8266-331968 kann man sehen dass der sensor kontinuierlich guten wifi empfang hat. 

Man kann auch sehen in https://www.madavi.de/sensor/graph.php?sensor=esp8266-331968-bme280#l_week dass der logger die environmental sensor daten zuverlaessig uebertraegt. d.h. deine beobachtung ist richtig, hier scheint es ein problem mit der kommunikation zu dem SDS011 sensor zu geben. 

was ueberraschend ist weil sich in dem code pfad nichts geaendert hat gegenueber der vorherigen version.

iceweasel1, 2019-11-14:
@dirkmueller Ups, sorry. War in der Annahme das dieser dann raus müsste, da dies eventuell eine höhere Prio als die Beta hätte. Na zum Glück falsch gedacht. Habe den Haken eben bei Auto-Update wieder rein und einen Neustart ausgeführt. Mal sehen wie es die kommenden Stunden sich macht. 

Ja bin derzeit verwundert. Mal schauen, vielleicht finden wir alle noch die Lösung.

iceweasel1, 2019-11-14:
Also, unter Beta NRZ-2019-126-B5" läuft es jetzt. Gespannt die 1. Messung erwartet, freunde aufgekommen, da gleich Werte angezeigt (Rede da ausschließlich gerade über die Übersicht, nicht Archiv) wurden bei PMs. Bei der 2. Messung doch in Ernüchterung umgeschlagen. Hatte mich zu früh gefreut. ;-)

Messung 3 soeben, wieder Werte dabei. Mal schauen wie es die kommenden Stunden ist, wenn ich sporadisch mal daraufschaue.

[image]

dirkmueller, 2019-11-14:
kannst du eine usb-serial consolen output capturen wenn das passiert und hier bitte anhaengen?

dokape, 2019-11-15:
spannend.
Im anderen Post #551 zeigen sich in den Beispielen folgendes:
Sind -- in den Messwerten, ist die Dauer der Messübertragung gering.
Sind Messwerte für PM vorhanden, liegt die Dauer der Messübertragung sehr sehr hoch über 10 000ms. 
Dein Aufruf der Webseite Luftdaten löst die IP-Adresse nicht auf. DNS-Fehler? 

Setze doch mal im Router die DNS-Adresse abweichend vom Internetanbieter auf 
1.1.1.1 (cloudflare DNS, extrem schnell) oder 8.8.8.8 für Google DNS. 

Einen generellen Fehler mit der Firmware 125-B1 kann ich nicht nachvollziehen, das läuft hier sauber. 
7545 Messungen in Folge, das sind rund 12,5 Tage.
Allerdings liegt bei mir auch die Dauer der Messübertragung bei in der Regel >2000ms. 
Deine 200-300ms bekomme ich nicht hin.

iceweasel1, 2019-11-15:
Ja spannend ist das Ganze. Der Aufruf der Webseite war nicht Luftdaten selbst, sondern wie in den #551 beschrieben ausschließlich die Maps/Karte gewesen. Meine in der #551 genannten Offtopics-Fehler-Hinweise sind mittlerweile, zumindest sehe ich es so, behoben worden.

Bislang war die Maps unter **https://deutschland.maps.luftdaten.info** zu erreichen. Das war auch bei der DNS-Fehlermeldung (siehe Screenshot) in der dort angezeigten URL eingetragen. Klickt man jetzt über die luftdaten.info-Webseite die Karte an, oder von wo anders, ist dort eine andere URL neuerdings eingetragen. Diese hier: **https://deutschland.maps.sensor.community**. Die "alte" URL geht zwar auch wieder, wie ich merke, ich vermute, dass ich vor 2 Wochen, wo ich bei #551 die Hinweise eingetragen hatte, einfach gerade da den Wechsel der Domain zufällig erwischt habe und die noch nicht Global durchging. Meine Vermutung zumindest.

Daher (denke ich) müsste ich nichts abändern an der DNS im Router, da die Fehler die ich bislang aufgezählt hatte (u. A. in #551), glücklicherweise nicht von mir stammen/ausgelöst wurden. :-) Belehrt mich was besserem, sollte ich falsch liegen.

Ich hatte mit der vorherigen Firmware keinerlei Probleme. Da hatte ich zwar noch nicht die ms sehen können, aber die Messungen gingen weit in den 5-stelligen Bereich, bis mal ein Neustart oder was anderes war. Auch gab es bis dahin noch nie, das in der Übersicht bei den PMs die Werte durch Bindestriche oder ähnlichem nicht zu sehen waren. Auch im Archiv sind keinerlei fehlende Werte bis dato zu erkennen gewesen. Daher hatte ich / bin bislang in der Annahme, das es an der neuen Firmware liegt. Denn genau seit dem Tag, seitdem die aufgespielt wurde, sind die genannten Probleme.

ricki-z, 2019-11-15:
Das Problem mit der Karte war (mal wieder) ein Problem beim Hoster. Der hat ein Update eingespielt und dabei wieder einen Fehler eingebaut, der schon mal behoben war.
Um die Karte irgendwie zur Verfügung zu stellen, haben wir als Notlösung auf die andere Domain gewechselt.
Diese Domain soll in naher Zukunft Luftdaten.info Teil ablösen, weil wir ja inzwischen mehr als nur Luftdaten speichern. Und mittlerweile auch recht international sind, wo dann der deutsche Name erklärungsbedürftig ist.

Ich würde die DNS-Einstellungen trotzdem überprüfen. Es kommt immer wieder vor, daß bei manchen Providern nur einer von zwei DNS-Servern funktioniert. Oder ein vor Jahren mal eingetragener DNS-Server inzwischen nicht mehr erreichbar ist. Wir hatten es schon mehr als einmal, daß genau damit ein Problem vorlag.
Und vor ein paar Wochen hatten Pi-Hole-Firewalls ein Problem, weil jemand eine völlig ungeeignetet Blacklist eingebaut hat. Eine eventuell vorhandene Firewall/Proxy sollte also auch kontrolliert werden. Die Update- und API-Server laufen mit der aktuellen Firmware unter der neuen Domain sensor.community.

iceweasel1, 2019-11-15:
> kannst du eine usb-serial consolen output capturen wenn das passiert und hier bitte anhaengen?

Werde ich über das Wochenende hoffentlich zeitlich umsetzen. Gebe bescheid, sobald getan.

iceweasel1, 2019-11-15:
**Zwischen-Update von gestern Abend bis jetzt** (Messung 438 von 16.50 Uhr)

[image]

Interessant ist im Archiv https://www.madavi.de/sensor/images/sensor-esp8266-331968-sds011-1-day.png , dass es wieder einen längeren, für die Grafik kurzen, Aussetzer anscheinend gab, siehe blauer Kreis. Wenn ich weiter reinzoome, kann ich keinen weiteren Aussetzer sehen.

[image]

Vergleiche ich das etwaige Zeitfenster des Aussetzers im Archiv parallel mit Grafana, zeigt es mir jedoch eine durchgehende Messung an. Oder interpretiere ich da was falsch?
https://maps.luftdaten.info/grafana/d/000000004/single-sensor-view?orgId=1&var-node=24640&from=1573808127958&to=1573814683365

reingezoomt: https://maps.luftdaten.info/grafana/d/000000004/single-sensor-view?orgId=1&var-node=24640&from=1573798552645&to=1573818366179&fullscreen&panelId=2

iceweasel1, 2019-11-15:
@ricki-z Ahh okay alles klar. Das ergibt alles Sinn und ist schlüssig. Ebenso wegen der Namensänderung bezüglich International. Danke für den Hinweis zum Thema Karte. 

Ich habe nach meinem Kommentar kurz vor 15 Uhr, wo @dokape das mit der DNS-Sache angab, dennoch einfach mal abgeändert. Nutze VDSL2 bei der Telekom (100/40), effektiv kommt etwas weniger rein.

[image]

Bislang war in der FritzBox die vom ISP zugewiesenen DNSen für IPv4 und 6, also auf automatisch eingestellt. Habe diese jetzt kurz nach 15 Uhr trotzdem mal auf die Empfehlung von @dokape auf die von Cloudflare geändert. 

[image]

dirkmueller, 2019-11-15:
> Vergleiche ich das etwaige Zeitfenster des Aussetzers im Archiv parallel mit Grafana, zeigt es mir jedoch eine durchgehende Messung an. Oder interpretiere ich da was falsch?

Ja, grafana macht per default lineare interpolation von messwerten. wenn madavi keine anzeigt gabs keine, und grafana macht das dann "glatt".

iceweasel1, 2019-11-15:
Okay. Das wusste ich jetzt nicht mit Grafana, das dies dann letztlich
glatt macht, wenn mal keine Daten vorhanden sind. :)

iceweasel1, 2019-11-23:
Guten Abend Zusammen,

erneutes Zwischenupdate: Nach 8 Tagen vom letzten Post-Update hier. Leider wird beim SDS gar nichts mehr protokolliert. Weder im Archiv, noch auf der Übersichtsseite wenn ich den Sensor über die IP ansteuere... Keine Messwerte mehr im Archiv und auf der Übersichtsseite sind die bereits genannten Bindestriche Dauergast. (Beide Screenshots Stand jetzt)

[image]
[image]

Nach dem Wechsel auf die Beta-Version, wie oben empfohlen wurde, ging es rund eineinhalb Tage gut. Es gab in der Übersicht (Screenshots habe ich) immer mal die Bindestriche und dann wieder die Messwerte, sowie im Archiv die Werte. Aber nach den knapp eineinhalb Tagen bis jetzt, konstant keinerlei Messwerte mehr beim Feinstaub.

An der Übertragung, wie bereits zuvor schon gesprochen, liegt es definitiv nicht. WLAN-Signal und die BME-Werte sind im Archiv ohne Unterbrechung bis heuer aufgezeichnet.

https://www.madavi.de/sensor/graph.php?sensor=esp8266-331968-bme280
https://www.madavi.de/sensor/signal.php?sensor=esp8266-331968

@dirkmueller kam leider am letzten Wochenende nicht dazu: https://github.com/opendata-stuttgart/sensors-software/issues/552#issuecomment-554411190 (usb-serial consolen output capturen)

dirkmueller, 2019-11-23:
Ich konnte das problem Mittlerweile nachvollziehen. Ich glaube er ist bereits gefixed. Leider ist der fix noch nicht im beta Channel verfügbar. @ricki-z wartet im Moment noch ab.

sanitariu, 2019-11-25:
I have similar problems with my sensors SDS011 and BME280. From some days both shows totally wrong information about Temperature,Humidity, PM10,PM2.5
How i can return to older version without dissasemble my station ?

iceweasel1, 2019-11-26:
@dirkmueller Kannst du es hier beschreiben was der Auslöser des Ganzen ist? Bin da sehr gespannt.

dirkmueller, 2019-11-26:
@iceweasel1 es gibt verschiedene Ursachen. Die ich hier vermute ist ein problem mit dem Interrupt Handling im arduino core, der dazu führt dass sds011 (oder jegliche andere serielle Kommunikation) für Minuten hängen bleibt.

Das war ein Fehler in dem C Compiler der verwendet wurde und ist mittlerweile im beta branch behoben.

iceweasel1, 2019-11-27:
@dirkmueller Ah, okay, alles klar. Habe eben wieder auf den Sensor geschaut und bemerkt, dass dieser jetzt bei Messung 223 ist. Muss also vor einigen Stunden selbst neu gestartet sein. Dürfte denke ich durch die neue Beta-Firmware sein? 

Sehe, dass ebenso bei der Firmware-Version oben nicht mehr B5 sondern B6 dabei steht, sowie unter den zwei PM-Werten ein neues Feld mit dem Sensor und dessen Parameter steht. Stehen sollte? Zumindest bei mir bleiben die Werte weiterhin leer. :D

Ein Blick, ob durch die Version B6 eventuell die auf der Übersicht nicht sichtbaren Werte trotzdem in das Archiv einfließen, blieb leider aus. Seit dem Neustart, scheint es weiterhin nicht zu Messen.

Ebenso ist im Archiv bei den PM-Werten jetzt auf der rechten Seite auf der Grafik die Einheiten von 0 bis 1.000 Meter (neu hinzugekommen?). Wie ist das zu Verstehen? Habe eben auf alte Screenshot geschaut. Da waren links und rechts die gleichen Zahlen-Skalen in der Regel gestanden.

[image]
[image]

dirkmueller, 2019-11-27:
@iceweasel1 ja, Interessanterweise ist -b6 nun schlechter in der direkten Kommunikation. Aber das ist gut, denn jetzt habe ich endlich das Problem gefunden. -b7 wird dieses Problem beheben.

iceweasel1, 2019-11-29:
@dirkmueller Hurra! B7 war ja schnell da. Hatte eben mal wieder darüber geschaut und mit erstaunen gesehen, dass die Werte wieder zu sehen sind. Gleich auf die Version oben geschaut und die 7 entdeckt. Im Archiv sehe ich ebenfalls bei den PMs wieder die Aufzeichnung. Das Update dürfte wohl gestern, der Aufzeichnung nach, in den Mittagsstunden rum, installiert worden sein?

Was ich im Archiv bei den PMs, bei der 24h- und Wochenansicht sehen kann, dass es anscheinend noch Aussetzer gibt. Nicht viele, aber solche die in der Grafik doch bemerkbar zu scheinen seinen. 

Ein Vergleich bei den BME280-Werten, die ja bislang immer aufgezeichnet wurden, zeigt mir, dass es zur selbigen Zeit ebenso bei den BME-Werten zu Aussetzern kommt. Alles was vor Donnerstag mittag bei den BME-Werten aufgezeichnet ist, scheint durchgehend, ohne erkennbaren Ausfall zu sein in der Grafik (wie es die ganze Zeit sonst war). WLAN-Qualität und Signal zeigt es aktuell bis jetzt, wie bereits vor Tagen und Wochen ebenso, ohne Aussetzer an.

Möchte nichts beschwören; eventuell möglich, dass das jetzige Beta B7 den gesamten Sensor (SDS samt BME) ab und an einen "Schluckauf" auslösen lässt?

[image]
[image]

dirkmueller, 2019-11-30:
@iceweasel1 ja, diese gelegentlichen Schlucklaufs sehe ich auch. Bei mir passiert das hauptsächlich zwischen 2 und 6 Uhr morgens, daher ist das debuggen nicht ganz so leicht..

Ich habe ein wenig Debugging Code eingebaut, Mal sehen ob er mir hilft dir Quelle zu finden.

Zumindest ist die heutige Version eine Verbesserung.

Adorfer, 2019-11-30:
Note: Es wäre sehr hilfreich (auch langfristig), wenn die exakte Firmwareversion im Titel angeführt wird. 
(Sofern das hier nicht als Sammelissue für "Probleme nach FW-Rollout" geplant ist)

ricki-z, 2019-11-30:
Kommentar zu den Grafiken auf Madavi.de:
Bei den Sensoren werden in den ersten Grafiken jeweils 24 Stunden angezeigt, beim Signal jedoch 3 Tage (= 72 Stunden). Dadurch können kurze Aussetzer, die in den Sensorgrafiken zu sehen sind, beim Signal verschwinden.

closed 2019-12-02 by ricki-z

dirkmueller, 2019-12-05:
Fyi die stabile Version wurde reverted. Die Beta Version sollte weiterhin den fix für dieses Problem enthalten.

commit 820122a9 (2019-11-27): Rewrote SDS handling from scratch (Fixes #552)

The yield() based decoding of SDS sensor input wasn't very
reliable, so occassionally the buffer wasn't decoded at the
time of the start of the read, which meant we were skipping
measurements. Now we're blocking until this has happened,
plus we align the decoding in the main loop to minimize blocking.

Also this saves quite a bit of code size, almost 400 bytes of flash.