opendata-stuttgart/sensors-software #557: Sensor sds011 - problems

sssimone, 2019-11-21:
Hello to all, i'm having some problems with sensor sds011.
Everything is regularly connected, cables are connected with lolin and it seems it works properly.
Anyways datas are too low:  i live in a city near street with car traffic, but all the day it detects only 0,1 or 0,2 for both pm10 and pm2.5
Here is link: http://www.madavi.de/sensor/graph.php?sensor=esp8266-3145321-sds011
Now is out of work, anyway you can see the past days....
Can someone help me about this matter?
May be the sds011 is broken or can i have made some mistakes during the software installation?
Thanks to all, ciao

dirkmueller, 2019-11-22:
@sssimone which firmware version are you using?

sssimone, 2019-11-22:
@dirkmueller here it is: NRZ-2019-126-B5

dirkmueller, 2019-11-22:
Did you check that the air suction nose is open? Do you maybe have too long pipe connected and the fan is unable to pull the air?

Try using it near a candle light that had just been blown out or similar. The values should spike high then.

Also, please ensure the sensor is oriented correctly. The fan unit just be the lowest (towards earth) and the circuit board highest (towards sky)

sssimone, 2019-11-22:
The nose seems to be open, the pipe is about 10cm...
I did follw your idea about candle and the sensor measured as follows:
pm2.5 3,6
pm10 13,1
now, it come back to 0,1 and it is 0,1.....too low agaiin

I just realized the fan is ok for some seconds, then it stops for about one minute and it starts again for few seconds... is it ok this way?
Ok to orient the sensor as per your instructions.

Thanks a lot for following my problem

dirkmueller, 2019-11-22:
Yes, the fan is only turned on at the measurement interval (which is by default 145s but you can lower it down to 30s if you want) for about 15s, and will then turn off again to increase longlivety.

dirkmueller, 2019-11-22:
So the air flow works as proven by the candle light experiment. Maybe remove the pipe for an experiment? Yet, the values are far too low. They should spike above 30 with ease (or higher).

Can you set the debug level to Max info and collect the serial output from the sensor while it is measuring and post it here?

sssimone, 2019-11-22:
Yes, i removed the pipe and seems to measure better than before, value about 30.
About your second question, i have to download again arduino and work on it.
I'm not so well served about this matter and a friend of mine is helping me.
So, i'm going to download arduino and let you know, anyway not today i think.

However, thank a lot again!
See you soon

sssimone, 2019-11-22:
Dirmueller, done what you told me:
[image]

sssimone, 2019-11-22:
Hello, done what you told me.

[image]

dirkmueller, 2019-11-23:
Great, thanks. So we can see that the sensor reports measurements, and the software can read them.

At this point in time I am confident that there is nothing wrong with the airrohr software. 

I suggest to double check the cabling. Contrary to what is intuition, the sensor needs 5v unlike any other sensor so that the fan and the laser gets enough voltage. I am guessing that the sensor is connected to 3.3v hence the fan doesn't provide enough suction.

See https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/Readme.md#anschluss-der-sensoren

sssimone, 2019-11-23:
Hello, i already changed the cables a couple of time and the sensor is going on to show datas between 0,0 and 0,1( inside home).
I confirm the cables are regulary connected.
One question: if i made wrong cabling connections, would the sensor have been reported the measures i linked yesterday? Or it would have been "out of order"?

About the voltage, when is out the sensor is connected with a power bank; when inside, like yesterday with the candle's smoke, it was connected with the pc.

dirkmueller, 2019-11-23:
So I tried with my dev setup SDS011. when I connect this setup's SDS011 to 3.3V, it appears to function properly from a  firmware perspective, so the serial communication works just fine, however the fan is less loud (it is a very old sensor) and the measurements are completely fluctuating (between < 1  and extremely high > 1000). So it is odd behaving, but not the "mostly flat 0.0" measurements that you're reporting. 

Maybe the power supply is not able to properly power the sensor, although total consumption should be within standard USB specification ( < 250 mAh). 

The other option is that for some reason your sensor is faulty. What is important to know is that the SDS011 is a factory calibrated digital sensor. With other words there is *nothing* in the software interface of the sensor where we can influence it or *reset* it or configure it differently. The only thing it does, when the fan is turned on, to send measurements (that it totally does by itself) once per second. We can not reprogram it or recalibrate it or update its firmware, for that matter. What we can see in the screen dump that you attached is that the communication works flawlessly, so there is no issue with the serial connection between the sensor and the microcontroller, it seems to work well.

At this point in time I'd say you have to experiment and figure out where the problem comes from. maybe the fan isn't perfectly connected or it isn't fully tightly sealed on the reflective silver/chrome piece. or try a different wall mounted power supply (like a regular micro usb phone charger ). or swap the SDS011 with a different unit.

sssimone, 2019-11-23:
I tried to change usb cable without success. I also tried to use a regular micro usb phone charger connected with a different wall mounted power supply. The sensor goes on showing 0,1....
So, i think the sensor is defective....
However, thanks a lot for your help and for the time you spent for me!!

dirkmueller, 2019-11-23:
If you feel comfortable with soldering you can open the sds011 and clean laser and sensor. It requires unsoldering and resoldering two flexible cables (the power supply for the laser). The resoldering is quite difficult, as the cable is extremely short and flexible. It will take more time than buying new - However it is also interesting ;-)

sssimone, 2019-11-23:
Thanks for your suggestion but i'm absolutely not able to do it :-(
I think i will buy a new sensor ;-)
Thanks again

dirkmueller, 2019-11-23:
Please let us know if a new sensor fixes the issue!

sanitariu, 2019-11-25:
Hello,
I have the save problem. I am using auto-update and current software version is: NRZ-2019-125-B1.
2-3 weeks before everything works like a charm for almost 2 years. I cleaned my station before 2 months and everything is ok. Also sensor BME280 gives too low results not normally.
Current temperature: 26 degree ( it is 3 degree now)
Also pressure and humidity is wrong.
PM2.5 shows 0.2 and PM10 shows also 0.2 which is not possible currently.
I think latest fixes did something wrong.
How i can return old version from before 3-4 months ?

dirkmueller, 2019-11-26:
@sanitariu can you please check the 'use beta version' checkbox in addition to the auto update flag and let us know if there's anything better with that?

dirkmueller, 2019-11-27:
@sssimone please hold off purchasing a new sensor until you tried firmware nrz-2019-126-b7 (or newer)

sanitariu, 2019-11-27:
Hello,
I am using beta: NRZ-2019-126-B6
My SDS011 is version 18-1-18(afdb).
I am using also BMP/E280.
Yes i am holding even going to the roof because both sensors stop working correctly.
Will wait for newer version.

dirkmueller, 2019-12-02:
@sssimone  @sanitariu please report back if you're on NRZ-2019-126-B9 or NRZ-2019-127 and still see issues.

0Raven, 2019-12-04:
Hello, same for me, NRZ-2019-127-1 and PM2.5 & PM10 shows 2 on sds011, near the street. It works correctly few days ago. I have auto-update on.

dirkmueller, 2019-12-07:
@0Raven is it back normal? we reverted to the previous firmware.

sanitariu, 2019-12-08:
Now I am with - NRZ-2019-125-B1
Seems like everything is ok

dirkmueller, 2019-12-15:
please test if https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B3-SDS011-fix/ fixes the unstable sds011 reads issue for you (this is unrelated to madavi - please file this as a separate issue).

sssimone, 2019-12-16:
@dirkmuller hello, i was holding everything since i was waiting for receiving the new sensor...sorry for my really late reply!
I now realized about some change: here is updated version NRZ-2019-128-B3. Is it the latest one?
Anyway, it still go on showing 0,0 both of pm
Moreover, each measurement reports an error....may be the sensor is really broken.
Anyway, these days i will try to check this matter, once i receive the new one i will try again, hoping to solve everything!

dirkmueller, 2019-12-19:
there is a newer one than B3 available as in the previous comment linked. it was proven to fix similar reports . If you could test it then that would iron out the last doubt that there is a software issue.

sssimone, 2019-12-22:
Hi DIrk, everything solved with the new sensor!
It arrived yesterday, i connected the cables and now it shows acceptable measurements.
So, the old sensor was broken!
Anyway, thanks again for your help!!

dirkmueller, 2019-12-22:
Thanks for the feedback! Glad to hear we found the cause.

closed 2019-12-22 by dirkmueller