opendata-stuttgart/sensors-software #562: Missing voices in italian configuration page.

stewecar, 2019-11-25:
Hello,
some of the "voices" in the sensor page are missing under 'configuration'.
For example 'load beta firmware' in italian is 'carica software beta' or flip display option is 'ruota il display di 180°"..are missing.
I could translate if you want and load in the beta branch.
s.

ricki-z, 2019-11-26:
Hi @stewecar it would be nice if you could add the translation to the beta branch.

stewecar, 2019-11-26:
sorry i am a newbie..should i modify the file in sensors-software/airrohr-firmware/intl_it.h or i have to upload a completely new one in the same directory? Thanks..

dirkmueller, 2019-11-26:
Just update the file in place and send a PR. Or if you're unsure, just paste the translations that should be added as a comment here.

stewecar, 2019-11-26:
thanks! i am updating it..

stewecar, 2019-11-26:
done! thanks!

dirkmueller, 2019-11-26:
@stewecar please change the pr to be submitted against the *beta* branch instead of *master*

stewecar, 2019-11-26:
sorry..done now..

stewecar, 2019-11-26:
there is a little change that i would like to make..i made a little error in a word..could i change it or is it a problem now? thanks!

dirkmueller, 2019-11-26:
I merged the typo fix.. should be all in #565

dirkmueller, 2019-11-27:
will be in 126-B7 or newer

commit 7ad4ee29 (2019-11-27): Update intl_it.h (Fixes #562)

I modified a few not available voices of the italian configuration's page.
another update of the italian translation
closed 2019-11-27 by dirkmueller