opendata-stuttgart/sensors-software #568: NRZ-2019-125-B1, lange Dauer Messübertragung

dokape, 2019-11-27:
Hi,

Der Wert der Messübertragung lag bei mir ürsprünglich bei Werten um 2000 ms.
Seit ein paar Tagen liegt er bei 40 000 bis 60 000ms. 

Konfiguration hier:
Über eine Repeater (TP-Link RE500 mit eigener Linkstrecke zu AP mit OpenWRT, RE355)
Über einen Router Linksys WRT3200ACM mit OpenWRT und Load-Balancing zu zwei Providern,
Unitymedia/Vodafone Cable und Telekom VDSL). Ich habe mit keiner anderen Anwendung Probleme mit dieser Konfiguration.

Die Messwerte werden an folgende API übertragen:
API Luftdaten.info  (HTTPS), ping: < 30ms
API Madavi.de  (HTTPS), ping: < 30ms

sowie im lokalen Netz auf einen RaspberryPi, (ping im Netzwerk < 10ms) der als eigene API die Daten ebenfalls nochmals aufnimmt. Der Raspi übernimmt mit #pihole als Adblocker und Trackingfilter die DNS-Auflösung. 
Als Script zur Lokalen Datenspeicherung dient das alte PHP-simple-Beispiel. 

Die neue Firmware 125-B1 läuft anscheinend sehr stabil seit rund 4 Wochen, wie der Zähler zeigt. 
 
Zur Ursachensuche der hohen Werte der Messübertragung habe ich Fragen: 
Wie wird die Dauer der Messübertragung ermittelt?
Ist es das Auslesen des SDS011?
Ist es die Übertragung zu Luftdaten.info/Madavi.de?
Ist es die Übertragung zum lokalen Raspi?

[image]

dokape, 2019-11-27:
Ich habe mal einige Tests gemacht:

Deaktiviere ich in der Konfiguration die API Luftdaten.de,
Sinken die messübertragungszeiten auf weniger als 200ms.
Eine Deaktivierung lediglich der HTTPS Option bringt keine Besserung.
Eine Deaktivierung lediglich von Madavi.de bringt ebenfalls keine Besserung, die Werte lagen immer noch über 10 000ms.

ricki-z, 2019-11-27:
Könnte man den Sensor (temporär) auf dem PiHole "freischalten"? Ich weiß aus der Vergangenheit, das dort api.luftdaten.info schon einmal auf einer Blacklist gelandet ist, weil für eine der Blacklists alle Domains genommen wurden, die aus Android-Apps heraus aufgerufen wurden. Also auch solche, von denen nur Daten geladen wurden, die für die Anwendung wirklich nötig waren. In unserem Fall haben die Apps einfach Daten von unserer API und unserem Archiv abgefragt.

dokape, 2019-11-27:
Freischalten ist kein Problem.
Ein deaktivieren des pihole brachte keine Besserung 
Die DNS-Anfragen  auf madavi.de sind im pihole log registriert und akzeptiert.
Für Luftdaten.de sehe ich keine Anfragen.
Sensor.cummunity Anfragen inkl. Subdomains sind erkennbar und ebenfalls frei mit antwortzeiten unter 1s, 
Würde geblockt, wären keine Sensorwerte bei Luftdaten ersichtlich. Hmm.. 

[image]

dirkmueller, 2019-11-27:
@ricki-z ich habe das gleiche problem, irgendwas ist mit luftdaten/sensor.community kaputt. submission times > 25s sind keine seltenheit.

ricki-z, 2019-11-28:
Irgend etwas blockiert gerade den Server alle 5 Minuten. Ich vermute ein Problem mit dem NAT/SAN beim Hoster.
Ob es eventuell einen anderen Grund gibt muss ich schauen, habe aber erst seit heute mittag wieder einen funktionierenden Rechner.

dirkmueller, 2019-11-29:
@dokape problem scheint geloest zu sein? Bei mir sind die werte wieder gut. ok zum schliessen?

dokape, 2019-12-01:
Ich beobachte noch.
Derzeit auf ca 2000ms runter. 
Was ich als subjektiv immer noch recht lange empfinde. 
Die paar Bytes sollten in 200ms Bereich übertragen sein. So subjektiv. (Upload hier um 30Mbit/sec)

dirkmueller, 2019-12-01:
das ist definitiv lange. zum vergleich habe ich ca 600ms fuer 3 server (madavi, sensor.community und fsapp)

dokape, 2019-12-02:
Gibt es eine Möglichkeit, App, Script, den Upload von Dummy-Daten nach Luftdaten.info 
zu generieren und zu schauen, die lange das dauert?

bertrik, 2019-12-02:
I'm running a Lora(TTN) to Luftdaten forwarder, that listens on an MQTT stream and uploads the data throught the Luftdaten HTTP API. What I saw in the past days was that the delivery took either a few 10s of ms (maybe 100 ms) or it got no response at all and timed out after 10 seconds (my own timeout). I suspect the high averages are caused by the transfers that eventually time out.
My forwarder software does not retry a failed upload (new data arrives within a few minutes anyway).

This is real data, I think it would pollute the database if you would send "dummy" data.

dirkmueller, 2019-12-02:
as far as I can tell this is a speciic issue with the luftdaten / sensorcommunity endpoint. and in my observation it is the SDS011 write not the environmental sensor (in my case bme280) that takes forever and in many cases times out

ricki-z, 2019-12-02:
I think its more the first request than the sds011 specific one. So if the environmental data would be send first this would take longer.

dokape, 2019-12-03:
I don't want to pollute the database. Therefore my question for dummy-data. 
I think to remember, data from not registered airrohr-IDs are uploaded but not stored. 
And perhaps for developing there could be a dummy-upload for dummy-data.

dokape, 2019-12-03:
mähhhh, i didn't want to close it. :-( 
cold fingers, fast klicking. wrong button. YEAH! ;-)

UliMy-B, 2020-06-06:
Ich kann das Problem bestätigen. Auch bei mir seit einiger Zeit sehr lange Meßübertragungszeit:
[image]
Auch scheinen viele Meßwerte nicht auf dem Server anzukommen. Die Kurver sind sehr lückenhaft:
[image]
Immer wieder resets, die die Anzahl der Messungen auf null setzen:
[image]