opendata-stuttgart/sensors-software #581: NRZ-2019-127-1 - SDS011 sensor issue.

Petarkir2000, 2019-12-04:
Hello guys,
We have a problem with the SDS011 sensor after last update-NRZ-2019-127-1.
The sensor cannot pas the check and is received error. The sensor is from 2017. See attache picture.
[image]
Before update the station was working properly. What would you advice?

dirkmueller, 2019-12-04:
Does it work after power off and on?

Petarkir2000, 2019-12-04:
No! I have tried that but without success.
The history of sensor was perfect: https://www.madavi.de/sensor/graph.php?sensor=esp8266-3389652-sds011

dirkmueller, 2019-12-04:
Hmm, thats unfortunate. Could you collect serial output of the sensor and paste it here (with max info)?

Petarkir2000, 2019-12-04:
if you tell me how :(

dirkmueller, 2019-12-04:
Depends on the Operating system. The esp8266 usb port is the serial, if you have an usb cable micro usb to usb you can connect it to a computer.

Then most os support the serial driver out of the box, so on windows you can use putty , on linux screen and in general the Arduino ide for example.

9600 baud 8N1

dirkmueller, 2019-12-04:
also if you have a chance to observe the sensor itself, is it running fan at all or the LED on the back of the SDS011 blinking?

what I can see from the screenshot that it seems the serial connection between sensor and mcu seems to be not working, I wonder if it isn't able to send commands or only not able to receive responses. 

I have seem this once so far where the LED on the sds011 sensor was constantly on (red color), but there a power cycle helped.

Petarkir2000, 2019-12-04:
Ok dirkmueller, I will try to collect the data from this sensor, but this is happening with all sensors from 2017, and there are at least 300 in Bulgaria ... We already have a lot of emails about this problem. Probably the older firmware does not answer the queries and gives an error, but in reality they work perfectly.

dirkmueller, 2019-12-04:
we'll revert to the previous version via OTA.

Petarkir2000, 2019-12-04:
I will request the problem sensor in the office and will send you the serial output shortly.
Thank you for understanding!

dirkmueller, 2019-12-04:
I wonder if this is also somewhat related to the esp8266 flavor. which board from https://frightanic.com/iot/comparison-of-esp8266-nodemcu-development-boards/ do you have?

Petarkir2000, 2019-12-04:
I will see actually all the hardware in 2 days and will answer you.

tmeidinger, 2019-12-05:
Same issue here with the SDS011. Updating to NRZ-2019-128-B2 seems to have it fixed.

ricki-z, 2019-12-05:
NRZ-2019-127-1 (intermediate release) and NRZ-2019-128-B2 are actually the same. So it was the restart that fixed this.

dirkmueller, 2019-12-05:
@tmeidinger how did you update? By checking the 'use beta channel' flag? And which sds sensor firmware version so you have (shown in the current values webui)

tmeidinger, 2019-12-06:
I just checked use beta channel. Firmware states 16-11-21(f90a).

dirkmueller, 2019-12-06:
@tmeidinger, ok so that sounds likely like a different issue as the original bug was about issues with sds sensors from 2017.

dirkmueller, 2019-12-10:
@Petarkir2000 did you manage to receive a sample that we can use for chasing this issue?

dirkmueller, 2019-12-15:
I believe I found a reason for this issue. please test if https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B3-SDS011-fix/

fixes that issue for you (remember to turn off auto update otherwise it will be replaced by the unfixed version on start).

Petarkir2000, 2019-12-16:
@dirkmueller we are testing this firmware now on 8 stations. Up to now - no issues.

dirkmueller, 2019-12-16:
@Petarkir2000 just to be sure -  the 128-b3-sds011 one?

That's great news then.

Petarkir2000, 2019-12-16:
Yes @dirkmueller. We use the firmware 128-b3-DSD-fix.
I found on my test station that when I use LCD screen 1602 (I2C: 0x27) and OpenSensamap the time in sending is increase to 16834 ms. Is that normal? Additionally no number in SDS011 Software version field. However no SDS011 errors.

dirkmueller, 2019-12-16:
please try if https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B3-SDS011-fix-long-delay/ recovers the software version field problem. 

the LCD screen should not have any effect on the sending time. the various sensor apis are having every once in a while issues, like right now madavi seems down (see #592  )

Petarkir2000, 2019-12-16:
Ooo, Yes, This version(NRZ-2019-128-B3-SDS011-fix-long-delay) is definitely better than the previous one. There is already an SDS011 number in the field (16-11-21(bf27) ).
And the response time is less than version NRZ-2019-125-B1. However we have to take into account that madavi.de has a problem.

dirkmueller, 2019-12-16:
Good news. okay, good that we solved this issue. should be in the next published beta then.

closed 2019-12-17 by dirkmueller

Petarkir2000, 2019-12-17:
Hello @dirkmueller I found that with the version NRZ-2019-128-B3-SDS011-fix-long-delay I cannot change the language. When I flashed latest_bg.bin I cannot change it on English for example. I did a clear flash with erase before.