opendata-stuttgart/sensors-software #584: Calibration/Adjustments required for higher altitude SDS011

erichiller, 2019-12-07:
I was reading through #171 and see that the recommended sensor is the SDS011. However I live at a higher altitude (around 1750 meters ASL) and the SDS011 datasheet says that it's operational pressure range is `86KPa~110KPa`. This poses a problem because my atmospheric pressure is around 83KPa. 
When using this sensor is there some way to calibrate it to make it work, or is this a hardware limitation?

stewecar, 2019-12-08:
It's definetely an hardware issue but i really don't know how much can affect your outputs..
You can skip to a PMS 7003 where It seems there are no limitations at all but i really don't know why..laser particle sensors work plus or minus in the same way..

dirkmueller, 2019-12-08:
@erichiller there is nothing customizeable from the published SDS011 sensor command interface, we can not tell it to do something else other than "do measure" or "don't measure and sleep". 

I would assume the issue is with the fan, the lower the pressure, the less airflow it will produce, and if there isn't (enough) airflow the sensor can not measure anything anymore.

dirkmueller, 2019-12-09:
see https://nettigo.pl/attachments/415 in case you're interested in the software protocol. I am closing as there is nothing we can do here..

closed 2019-12-09 by dirkmueller