opendata-stuttgart/sensors-software #585: Compiling errors when building firmware: wrong SoftwareSerial signature

chribben, 2019-12-08:
I've followed the [instructions](https://github.com/opendata-stuttgart/sensors-software/tree/master/airrohr-firmware#ben%C3%B6tigte-software-in-klammern-getestete-version-und-die-art-der-lizenz) concerning which libraries to include but when verifying/compiling in Arduino IDE I encounter the following error:

> airrohr-firmware:392:64: error: no matching function for call to 'SoftwareSerial::SoftwareSerial(const uint8_t&, const uint8_t&, bool, int)'
> 
>  SoftwareSerial serialSDS(PM_SERIAL_RX, PM_SERIAL_TX, false, 128);
>                                                                 ^
> c:\dev\sensors-software\airrohr-firmware\airrohr-firmware.ino:392:64: note: candidates are:
> 
> In file included from c:\dev\sensors-software\airrohr-firmware\airrohr-firmware.ino:135:0:
> 
> C:\Users\....\Documents\Arduino\libraries\SoftwareSerial\src/SoftwareSerial.h:94:5: note: SoftwareSerial::SoftwareSerial(int8_t, int8_t, bool)
> 
>   SoftwareSerial(int8_t rxPin, int8_t txPin = -1, bool invert = false);
>   ^
> 
> C:\Users\.....\Documents\Arduino\libraries\SoftwareSerial\src/SoftwareSerial.h:94:5: note:   candidate expects 3 arguments, 4 provided
> 
> C:\Users\.....\Documents\Arduino\libraries\SoftwareSerial\src/SoftwareSerial.h:90:5: note: SoftwareSerial::SoftwareSerial()
>      SoftwareSerial();
>      ^
> C:\Users\......\Documents\Arduino\libraries\SoftwareSerial\src/SoftwareSerial.h:90:5: note:   candidate expects 0 arguments, 4 provided

dirkmueller, 2019-12-09:
there must be something modified in your sources. the line 392 does not have this error in git. please double check that you don't have local modifications. 

you can check that (unmodified) git sources are building:

https://github.com/opendata-stuttgart/sensors-software/blob/master/readme.md

dirkmueller, 2019-12-10:
please let us know if updating to pristine git state resolved the issue for you.

chribben, 2019-12-10:
It appeared my local master was behind 138 commits (!)
After updating there was still one compile error namely that it couldn't find Adafruit_SHT31.h. After installing corresponding library it worked. Thanks!

closed 2019-12-10 by chribben