opendata-stuttgart/sensors-software #586: Keine PM-Werte mehr - NRZ-2019-128-B2

iceweasel1, 2019-12-09:
Abend Zusammen,

ich knüpfe an den Thread #552 an. Mein letzer Post vor 10 Tagen in dem genannten Thread hatte noch die Beta 126-B6. Der Sprung zum derzeit beibehaltenen 128-B2 ging flott. 1 oder 2 Tage hatte ich glaube ich die 127 gesehen. Da waren alle Werte zu sehen. Bei der 128-B2 waren die PM-Werte anfänglich kontinuierlich zu sehen. 

Seit Samstag, den 07. Dez. um 00.26 Uhr; dort wurde zuletzt aufgezeichnet, danach bis jetzt nichts mehr. Durchgehend die Bindestriche. Gestern um 08.02 Uhr kam eure automatisierte Mail, dass der seit 24 Std keine Daten mehr gesendet wurden. Wie bei #552 ist der Sensor ununterbrochen am Netz. WLAN und BME-Daten sind ununterbrochen im Archiv zu finden.

Screen von eben.

[image]

https://www.madavi.de/sensor/graph.php?sensor=esp8266-331968-sds011#l_week

Da das alte Issue geschlossen wurden, habe ich das neue eröffnet. Mit der Note die @Adorfer in die Runde schmiss, habe ich die Firmware-Version gleich in den Titel.

@dirkmueller du hattest vor 4 Tagen geschrieben, dass die stabile Version reverted wurde. Heißt das, ich könnte von der Beta in die normale wieder wechseln und hätte die Werte wieder? ;-)

dirkmueller, 2019-12-09:
von beta auf normale wechseln ergibt die version NRZ-2019-125-B1, aka die vom 31. Oktober 2019. diese sollte funktionieren (und so sollte auch die aktuelle beta)

es waere natuerlich viel spannender herauszufinden wo das problem liegt.. ich vermute es geht wieder nach einem power cycle des nodes?

iceweasel1, 2019-12-10:
Danke für den Hinweis.

Also aktuell Neustart zeigt nach 3 Messungen, bislang keine einzige Messung an. Werde es den Tag über beobachten ob es Mal aufzeichnet bzw. anzeigt. Melde mich dann wieder.

[image]

dirkmueller, 2019-12-10:
Neustart über die webui oder tatsächlich stromlos gemacht?

Ich vermute der Sensor hat sich aufgehängt, ein mcu restart alleine würde nicht helfen

iceweasel1, 2019-12-14:
Kam erst gestern Abend dazu. Hatte gestern Spätnachmittag das Netzteil aus der Steckdose. Paar Minuten gewartet und danach wieder eingesteckt. Die Messung jetzt ist 552. Die Nacht und vergangene Stunden, bei der Nachschau auf die Übersicht war ein 50/50-Spiel. Mal die Bindestriche, mal Werte wie jetzt z. B.

Im Archiv sind kleine Aussetzer zu sehen. Das deckt sich nicht ganz mit den in der Übersicht fehlenden Werten. 

Zumindest aber mal längere Messwerte am Stück :)

[image]

iceweasel1, 2019-12-14:
**Offtopic-Frage:** Ist das normal, das im Archiv die Monatsübersicht beim BME bei der Temperatur komplett leer bleibt?

[image]

dirkmueller, 2019-12-15:
please test if https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B3-SDS011-fix/ fixes the unstable sds011 reads issue for you (this is unrelated to madavi - please file this as a separate issue).

iceweasel1, 2019-12-18:
Gerade die automatisierte Mail erhalten, das seit ü. 24h keine Daten mehr eingegangen wären... PM-Werte ja, aber die BME-Werte etc. sind nicht davon betroffen. Diese kann ich im Archiv <24h bis sehen.

[image]

Aktuelle Messung
[image]

dirkmueller, 2019-12-18:
Which firmware is that? The one from the link in my previous comment?

dirkmueller, 2019-12-18:
Does this recover after a restart?

iceweasel1, 2019-12-18:
> Which firmware is that? The one from the link in my previous comment?

Korrekt. Nach dem Einspielen der "de".bin, Neustart gemacht in Form von Stromlos und danach wieder rein.

dirkmueller, 2019-12-19:
geht es wieder nach dem man den node neugestartet hat?

dirkmueller, 2020-01-17:
@iceweasel1 any update? does power off/on fix the problem?

iceweasel1, 2020-01-21:
> geht es wieder nach dem man den node neugestartet hat?

Sorry, konnte mich nicht früher zurückmelden.

Du meinst mit node neustarten, Strom komplett weg, kurz warten und dann wieder Strom ran?

dirkmueller, 2020-01-22:
Ja, genau. Manchmal hängt sich doch der sds011 auf und die rote led ist Dauer an.

iceweasel1, 2020-01-22:
@dirkmueller Alles klar, Danke dir. Habe es heute Mittag ausgeführt. Weg vom Strom, kurz gewartet, danach wieder angeschlossen. Aktuell sieht das Ganze so aus. Das Thema mal stromlos zu machen habe ich ja schon öfters getan. Zuletzt zu Jahresbeginn. Da hielt das Ganze rund eine Woche, bis eben bis heute dann kontinuierlich keine einzige PM-Messung ging. Alles andere wie Luftdruck, Feuchte und Temperatur geht ja ununterbrochen ohne ein einziges Problem, seit bestehen des Sensor. Hoffe, dass halt nicht wieder in spätestens einer Woche der Node durch das Strom-Thema neu gestartet werden muss.

[image]
[image]
[image]
[image]

dirkmueller, 2020-01-22:
Ich konnte ein wahrscheinlich ähnliches Problem vor ein paar tagen nachvollziehen. Wenn ein reboot des devices passiert ( z.b. wegen eines OTA) während der Sensor angeschaltet ist stürzt der Sensor manchmal ab. Das erkennt man an einer dauerhaft leuchtenden led und daran dass er jegliche Kommunikation einstellt und erst durch stromlos wiederbelebt werden kann. 

Ich habe das erstmal dadurch gelöst daß jetzt OTA nur noch gemacht wird wenn der Sensor gerade aus ist . Im diese Änderung auszurollen brauchen wir aber noch ein OTA, d.h. einmal kann es noch, danach sollte das Problem erstmal weg sein.

iceweasel1, 2020-01-22:
Interessant zu wissen. Danke für den Hinweis. Kleine Sache mit großer Wirkung. Sehe es bei mir leider nicht, da das Rohr in luftiger Höhe hängt. Alles klar, dann weis ich bescheid und nehme diese danach nochmal vom Strom etc. Hast du schon einen etwaigen Zeitraum wann das ausgespielt wird? Nächste(n) Tage oder Woche(n)?

93548, 2020-01-23:
Mein Gerät sendet keine PM-Daten mehr, wohl aber Feuchte und Temp. Rote LED ist auf Dauer an. Strom abstecken, warten, anstecken hilft nicht. Firmware ist NRZ-2020-129
Was kann ich tun?

iceweasel1, 2020-01-26:
Hab gerade so spontan wieder auf die Übersicht geschaut und gesehen, dass die PM-Werte (heute das erste Mal nach letztem Neustart) weg sind. Gleich in die Geräteübersicht geschaut und bei "Last OTA" gesehen das vor 4 min eins war. Gleich Strom weg, kurz gewartet und wieder ans Netz. 

Sollte jetzt ja erledigt sein. 😊

iceweasel1, 2020-01-28:
Ich bin irritiert. Stunden nach meinem Post vorgestern hat es leider wieder angefangen. Gegen Abend als ich nachschaute, waren wieder die Bindestriche zu sehen bei den PM-Werten. Im Gerätestatus stand, das rund 4 Stunden zuvor eine OTA ausgeführt wurde. Vorgestern sowie gestern und heute gab es laut Gerätestatus, wo ich in den drei Tagen immer wieder hineinschaute, zu unterschiedlichen Zeiten mehrere OTAs bei "Last OTA" mit den Zeitenwerten. So oft komme ich da aber nicht hinterher den Sensor dann immer wieder kurz vom Strom zu trennen etc.

Was sich aber null mehr ändert, ist die Firmwarebezeichnung. Die bleibt bei NRZ-2020-129 mit 7. Januar.

Hab ich da einen Denkfehler? Dachte die OTA tut i. d. R. vorlaufend bereitstehende Updates einspielen? Außer der Name der Updates blieb jetzt seit 07. Januar gleich? :D

[image]
[image]
[image]

dirkmueller, 2020-01-28:
'Last OTA' bezieht sich auf den Zeitpunkt wann zuletzt nach einem Update gesucht wurde, nicht wann das letzte Mal ein Update durchgeführt wurde, das kann man an der 'Uptime' u.U. erkennen zusammen ob das neu Start als 'system restart' gelistet wurde.

iceweasel1, 2020-01-28:
Ahhh okay. Da war mein Denkfehler. Danke dir. Der externe Neustart durch den Strom war ja vor 2 Tagen. Diesmal hielt es nicht wie die vergangenen Male mehrere Tage durch. Waren jetzt nur 24h wo aufgezeichnet wurde. Automatische Mail (keine Daten 24h) von Euch kam vor paar Minuten rein. Mhh.

bdcaf, 2020-02-03:
Ich glaube ich habe  dasselbe Problem. Vor ein paar Tagen kam das Mail. Heute hab' ich alles mehmals zerlegt und neu zusammengesteckt - leider ohne erfolg.
Dachte erst der Sensor sei schon kaputt (Alter ca. 30 Tage). 
Mir fällt keine leuchtende LED auf - wo sollte die sein?  Wie lange muss vom Strom getrennt werden? Bei mir war beim zusammenstecken ca. 5 Minuten die anscheinend nicht reichen.

dirkmueller, 2020-02-03:
[image]

Die Diode markiert mit "D23" sollte ca 1 mal die sekunde fuer einen kurzen augenblick blinken waehrend der sds011 sensor misst.  leuchtet diese dauerhaft, ist der sensor abgestuerzt.

dirkmueller, 2020-02-03:
Bei meinen Experimenten reicht eine trennung vom strom fuer ca 30 sekunden aus um den sensor zu recovern. 

sollte die LED in ca 1* sekunde interval blinken fuer ca 20 mal alle messinterval (also ca alle 2.5 minuten standardmaessig) aber trotzdem keine werte auf der website erscheinen ist das problem ein anderes. in dem fall bitte mal die serielle debug ausgabe posten.

93548, 2020-02-04:
Bei mir lag das Dauer-Rot der LED an defekten Steckern der Verbindungskabel. Finde die Stecker recht fragil. Habe sie ausgewechselt und mit Heißkleber an den Platinen fixiert. Danach sendet er wieder. Ggf. kann man auch löten.

bdcaf, 2020-02-14:
Danke für die Tipps. Also es ein paar Messungen er nun schon erfolgreich gemacht. Habe die Drähte verlötet - nicht sonderlich professionell, aber ich hoffe mit viel Heißkleber halten sie.

Mir war erst etwas schummrig weil 30s nicht reichten. Aber scheint ein paar stromlose Tage taten den Trick.