opendata-stuttgart/sensors-software #587: compile Error Master NRZ-2019-127

DigiE2, 2019-12-09:
nach dem update/pull auf NRZ-2019-127 master
(inkl. Update to Arduino Core 2.6.2)  hatte ich 2 compile erros:
* missing Adafruit_SHT31.h - nach dem installieren der Version 1.5.1 -> OK  
  fehlt evtl. in [https://github.com/opendata-stuttgart/sensors-software/tree/master/airrohr-firmware#verwendete-bibliotheken-f%C3%BCr-esp8266]
* airrohr-firmware:3701:59: error: 'empty_String' was not declared in this scope
    while (line_count < 3) { display_lines[line_count++] = empty_String; }
OK ist: `while (line_count < 3) { display_lines[line_count++] = **emptyString**; }`

Ist das nur bei mir so?
Thx Jens

dirkmueller, 2019-12-09:
die zeilennummer passt nicht auf master: https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/airrohr-firmware.ino#L3701

empty_String ist in der tat falsch, es sollte emptyString sein. ich finde das aber nicht im git. 

zu dem SHT thema gibts bereits ein PR.

dirkmueller, 2019-12-09:
der CI build test succeeded (es gibt dazu travis-ci marker in der readme datei, den folgenden link mit einem webbrowser oeffnen:

https://github.com/opendata-stuttgart/sensors-software/blob/master/readme.md

closed 2019-12-09 by dirkmueller