opendata-stuttgart/sensors-software #588: sending Data via TTN ? -- no wifi, no cellular ...

jensileinchen, 2019-12-10:
Dear community, i would kindly ask for some help in my ideas .... 

>> as we have in our village the possibility to connect any node to the TheThingsNetwork  ( 2 Lorawan- Gateways are available) i had the idea to move the SDS011 with BME280 (and/or DHT11) to an ESP32 board like Paxcounter (TTGO) or TTGO-BEAM ..... We have permanent powersupply available, but no "open" wifi in near range to reach. Also no possibility to add a wifi router with "Freifunk" --- Also no private useres are willing to open their networt to send data from sensor(s) ....

ricki-z, 2019-12-11:
The ESP32 code parts are the preparation for the TTN support.
There are some implementations out there that use an ESP32 and are forwarding the measurements to our database. I.e. https://github.com/verschwoerhaus/ttn-ulm-muecke
There is an option for ESP8266 but this will left only 3 useable pins, limiting the nodes to a SDS011 and a DHT22 (or other one wire temperature/humidity sensor).

Nordrunner01, 2019-12-11:
Hallo.

Meaning like this?: https://github.com/urs8000/particulate-matter_ESP32_SDS011_BME280_DHT22

Gruß

jensileinchen, 2019-12-12:
> 
> 
> Hallo.
> 
> Meaning like this?: https://github.com/urs8000/particulate-matter_ESP32_SDS011_BME280_DHT22
> 
> Gruß

yeppp ... just like this ..... i would prefer a "ready to use" version with TTGO board or TTGO-Beam (to send paxcounter data also --- but this version is similar to that what i am looking for .....

tho-wa, 2019-12-17:
I'm using sucessful https://github.com/verschwoerhaus/ttn-ulm-feinstaub with TheThings Uno.