opendata-stuttgart/sensors-software #590: Ping geht, Webseite und Datenübertragung tot

msebald, 2019-12-14:
Hallo,
habe ein seltsames Problem mit meinem Luftdaten Sensor. Ping geht (dh das Ding verbindet sich auch mit dem WLAN), aber die Webseite und auch die Übertragung der Daten ins Internet ist tot. Das Thema ist seit rund 10 Tagen der Fall, wurde durch die Statusmail darauf aufmerksam. Mehrfach neu starten (vom Strom trennen, auch mal länger) hat nichts gebracht.
Nun frage ich mich, was hier passiert ist, einfach so, und wie ich es gelöst bekomme. Etwa durch neu flashen?
Danke und Grüße,
Martin

dirkmueller, 2019-12-15:
Welche firmware version?

msebald, 2019-12-15:
Tja, wenn ich das nur wüsste. Komme ja nicht mehr drauf. An sich hat sich aktualisiert sich die Box doch automatisch? Falls nicht, ist es die vom Einrichtungszeitpunkt.

dirkmueller, 2019-12-15:
Wenn es seit 10 tagen ist dann müsste es die stable version NRZ-2019-127-1 sein..

msebald, 2019-12-15:
Inzwischen bissle länger, aber ja, vermutlich. Letzte Meldung wohl um am 03.12.2019 um 15:42.

dirkmueller, 2019-12-15:
Wenn es das bereits bekannte problem ist, müsste sich der sensor nach ca 3 Minuten nach Strom an wieder über die Webseite erreichbar werden. Wenn es das nicht ist, muss man leider eine Firmware von hand flashen.

Am besten wäre es wenn https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B3-SDS011-fix/ das Problem löst.

Ansonsten müsste man händisch die Firmware von  https://firmware.sensor.community/airrohr/update/ installieren. Beide müssten gehen, ansonsten gibt es ein neues noch unbekanntes problem.

dirkmueller, 2019-12-15:
Welche Sensoren sind an dem sensor angeschlossen?

msebald, 2019-12-15:
Habe ich es richtig verstanden, dass es diesbezüglich ein Problem gibt und das eventuell wieder passieren könnte? Gut, zur Not krabbel ich nachher halb auf's Dach und versuche den Sensor einzuholen und dann neu zu flashen. Weiß gar nicht mehr, was da so alles einzustellen ist, die Settings werden wohl weg sein. Auch will ich eigentlich keine neue ID bekommen.

> Welche Sensoren sind an dem sensor angeschlossen?

SDS011, DHT22 und BME280.

dirkmueller, 2019-12-15:
ID bleibt erhalten, wenn man kein erase_flash durchfuehrt (was nicht notwendig ist) bleiben auch die settings erhalten. 

man kann mit einem kommando wie z.B. 

```
esptool.py  --chip auto --port /dev/usbport --baud 460800 write_flash -fm dio 0x00000  latest_XX.bin
```

in place die firmware tauschen.

dirkmueller, 2019-12-15:
> Habe ich es richtig verstanden, dass es diesbezüglich ein Problem gibt und das eventuell wieder passieren könnte? 

Es sollte nie in der stabilen firmware passieren. In den Beta versionen hatten wir immer wieder mal regressions, daher ist sie aber auch als beta markiert. 

Am 3. Dezember wurde eine neue firmware im "stabilen" channel ausgerollt die u.A. eine neue arduino core version mit neuem WiFi Stack enthielt. Da gab es betroffene geraete die nicht mehr richtig damit funktioniert haben. Wir haben sie nach zwei tagen wieder auf den alten stand zurueck gestellt. Leider konnten nicht alle geraete automatisch mehr downgraden. Warum, wissen wir bisher nicht. 

insofern waere die serielle ausgabe des sensors sehr nuetzlich, um eine ursache eingrenzen zu koennen.

msebald, 2019-12-15:
Danke für die Unterstützung! Flashen hat problemlos geklappt und ich musste nicht einmal das Dachfenster aufmachen. Flashen ohne Pins kurzschließen und dergleichen, genial. Wusste nicht mehr, wie das bei dem Sensor war.

Webseite geht wieder und es werden auch Werte ermittelt. Und sicherlich auch übertragen (nicht gecheckt, aber das aber im Laufe des Abends oder morgen früh beobachten.

dirkmueller, 2019-12-15:
Welche Firmware Version läuft nun?

msebald, 2019-12-15:
> Welche Firmware Version läuft nun?

NRZ-2019-125-B1

dirkmueller, 2019-12-16:
Okay, das ist schonmal gut zu wissen. Ich wuerde allerdings gerne verstehen was das problem ist, damit wir es fuer das naechste OTA update beheben koennen. 

Koennen wir bitte die firmware von https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B3-SDS011-fix-long-delay/ testen? bitte vorher "updates" in der /config seite abschalten und abspeichern, und dann diese firmware flashen. 

geht diese auch? das waere dann die die im naechsten BETA update released werden wuerde.

BeNeDeLuX, 2019-12-27:
**Zur Info:**

Ich habe genau das selbe Problem (Ping geht, Webseite und Datenübertragung tot) und bin dann wohl auch einer der betroffenen vom OTA Update. Seit dem 03.12.2019 um 02:00 Uhr bekomme ich keine Daten mehr. Ist sehr gut in Grafana ersichtlich.

Werde in dem Fall abbauen und neu flashen. Meine genaue Version kann ich nicht mehr sagen und nicht mehr einsehen. Bin Ende Oktober mit dem Projekt gestartet und habe zu der Zeit die aktuelle stable Version genommen (falls die Info weiterhilft)

msebald, 2019-12-27:
@dirkmueller sorry, dass ich mich seither nicht gemeldet habe, war im Vorweihnachtsstress. Soll ich die Firmware noch testen oder inzwischen eine andere? Dann mache ich das nun zwischen den Jahren sehr gerne!

@BeNeDeLuX Abbauen ist ggf nicht nötig, wenn Du Dich mit nem Notebook hinbewegen kannst und das USB Kabel vom Sensor statt ins Netzteil ins Notebook steckst. Geflasht ist dann binnen Sekunden, da kannst solange sogar das Notebook festhalten.

erhapp, 2019-12-27:
Ich habe die gleichen Probleme (ping und kein Webserver).

Firmware: NRZ-2019-125-B1

dirkmueller, 2019-12-28:
@msebald @erhapp bitte die im vorherigen Kommentar verlinkte Firmware testen und berichten ob es danach geht!

benben, 2019-12-30:
`NRZ-2019-124-B10` und `NRZ-2019-125-B1` haben bei mir ähnliche Probleme verursacht.
`NRZ-2019-128-B4` scheint zu funktionieren. Danke!

msebald, 2020-01-01:
@dirkmueller Flashen auf NRZ-2019-128-B3 hat geklappt. Das Device liefert Werte und die Webseite ist erreichbar. (Habe das heute endlich gemacht nachdem der Sensor heute Nacht auf einmal weg war und bis eben nicht wieder kam. Schade, einmal im Jahr was zu messen und dann sowas. ;-) )

Wäre NRZ-2019-128-B4 nun doch besser gewesen als Test? Das hat ja @benben nun geflasht.

dirkmueller, 2020-01-01:
@msebald in der Website bitte Mal 'update' und 'beta kanal verwenden' anmachen und dann abspeichern. Interessant wäre dann zu wissen ob es dann auf -B6 updated und noch immer alles geht

msebald, 2020-01-01:
@dirkmueller Habe ich aktiviert und gespeichert. Nach wie vor B3, aber mal ein wenig abwarten. Schaue 1-2 Mal am Tag drauf und melde mich dann.

dirkmueller, 2020-01-01:
Das kann eigentlich nicht sein, wenn die Firmware zu alt ist updated er sofort..

msebald, 2020-01-01:
@dirkmueller Dann klappt's nicht. Folgendes ist aktiv: "Auto Update" und "Lade Beta Versionen" unter "Weitere Einstellungen". Nach wie vor ist das Device auf B3 und auch online.

benben, 2020-01-02:
Habe auch vor ein paar Tagen Auto Update und Beta Versionen angeschaltet (so habe ich auf B4 geupdatet). Bei mir hat es irgendwann auf `NRZ-2019-128-B6` geupdatet und funktioniert immernoch alles. Gebt Bescheid, falls ich irgendwie helfen kann.

dirkmueller, 2020-01-02:
@msebald kannst du die serielle ausgabe des sensors capturen? Ich vermute es gibt ein problem mit NTP oder mit der system zeit allgemein. am besten vorher auf "max info" setzen und abspeichern.

wenn die systemuhrzeit komplett kaputt ist (also aelter als das jahr des firmware builds, was hier 2019 ist), dann schaltet er HTTPS validation ab, um zumindest ein OTA machen zu koennen. wenn sie nicht ganz kaputt ist dann enforced er validation. ich vermute dein sensor laeuft mit ein paar wochen/monaten falscher uhrzeit und kann daher nicht updaten. 

ich hab leider keine gute idee wie man das fixen kann ohne HTTPS abzuschalten  (was als sicherheitsfeature eigentlich notwendig ist)

msebald, 2020-01-02:
Also Anfang Dezember hat er ja noch aktualisiert. Gut, da war das Jahr dann eventuell noch das gleiche, wenn Du mit Deiner Vermutung richtig liegst.

Serielle Ausgabe? Ich muss also wieder in die Dachkammer krabbeln? Also gut...

Generell sinnvoll wäre sicherlich, wenn man die serielle Console so tasmotamäßig auf der Webseite sehen könnte. Oder zumindest das aktuelle Datum/Uhrzeit im Webseitenheader oder bei "Aktuelle Werte". Oder allgemein eben die komplette Netzwerkconfig, wo ja per DHCP auch NTP mitkommen sollte. Oder ist NTP fest eingestellt?

dirkmueller, 2020-01-02:
@msebald ja, den seriellen output auf der website zu mirroren halte ich auch fuer eine gute idee. bisher waren die meisten probleme wo serieller output notwendig ist aber wifi related, und da hilfts dann auch nciht..

fuer OTA und NTP/date+time debugging gibts mittlerweile in neueren versionen eine eigene website (http://airror-xxxxxx.local/status). in der hoffnung dass das problem auch in der neuen version auftritt koennte man haendisch eine neue firmware flashen:

https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B7/

msebald, 2020-01-03:
Ok, seriell ausgelesen (für alle anderen: z.B. mit dem Seriellen Monitor der Arduino IDE und 9600 Baud), folgendes kam raus:

```
Airrohr: NRZ-2019-128-B3/DE
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to xxx
.......
WiFi connected, IP is: x.x.x.x
Starting Webserver... x.x.x.x

ChipId: 3912888
Read SDS...: 16-11-21(063d)
Stopping SDS011...
Read DHT...
Read BMP280/BME280...
Trying BMP280/BME280 sensor on 76 ... found
Send to :
sensor.community
Madavi.de
----
Auto-Update active...
ws: root ...
PM10:  22.90
PM2.5: 17.17
----
## Sending to sensor.community - SDS011
Time incorrect; Disabling CA verification.
Succeeded - api.sensor.community
Temperature (°C): 4.00
Humidity (%): 99.90
----
## Sending to sensor.community - DHT22
Time incorrect; Disabling CA verification.
Succeeded - api.sensor.community
Temperature (°C): 4.82
Pressure (hPa): 98842.91
Humidity (%): 100.00
----
## Sending to sensor.community - BMP/E280
Time incorrect; Disabling CA verification.
Request failed with error: 400
Details:["sensor could not be selected."]
## Sending to madavi.de: 
Time incorrect; Disabling CA verification.
Succeeded - api-rrd.madavi.de
## Sending to opensensemap: 
Time incorrect; Disabling CA verification.
Succeeded - ingress.opensensemap.org
Time for Sending (ms): 351
```

@dirkmueller Deine Vermutung mit der Zeit war also korrekt. Da stimmt was nicht. Nur was fange ich damit nun an bzw kann es beheben? Weiß nicht, was für ein Zeitserver gesetzt ist (statisch oder per DHCP) und was es damit ggf für ein Thema gibt.

Grundsätzlich wie gesagt fände ich den seriellen Output auf der Webseite sehr sinnvoll, auch, wenn es natürlich nichts bringt, wenn die Webseite nicht läuft. Auch wären die Settings (DHCP, NTP, Datum/Uhrzeit...) auf der Webseite supersinnvoll. Das frisst ja kein Brot und sind nur auf einer der Seiten ein paar zusätzliche Zeilen. Auch toll wäre ein Updater via Web (ich danke da wieder an Tasmota). Aber k.A., ob so eine Funktion noch auf das Device passt.

Btw, soll ich nun auf die von Dir genannte NRZ-2019-128-B7 updaten? Laptop liegt noch in der Dachkammer.

dirkmueller, 2020-01-03:
@msebald ja, bitte auf -B7 updaten. die NTP server werden im moment nicht von dhcp uebernommen (but, not a feature) sondern sind statisch. d.h. netzwerkmaessig kommt er nicht an sie ran. 

der OTA updater sollte trotzdem gehen, aber tut er nicht. das koennen wir mit -B7 debuggen, weil die fehler dort in der webui reported werden.

msebald, 2020-01-03:
Ok, manuelles Update auf NRZ-2019-128-B7 (von o.g. Link) gemacht. Folgendes kam nun:

```
Airrohr: NRZ-2019-128-B7/DE
mounting FS...
opened config file...
parsed json...
Rewriting old config from: NRZ-2019-128-B3
Saving config...
Config written successfully.
output debug text to displays...
Connecting to xxx
.....
WiFi connected, IP is: x.x.x.x
Starting Webserver... x.x.x.x

ChipId: 3912888
Read SDS...: 
Stopping SDS011...
Read DHT...
Read BMP280/BME280...
Trying BMP280/BME280 sensor on 76 ... found
Send to :
sensor.community
Madavi.de
----
Auto-Update active...
SNTP sync finished: Fri Jan  3 14:39:56 2020

twoStageOTAUpdate
Update md5: 40099cbc1cb8cf9aa792a0e7939584f4
Sketch md5: d63cbaf289b93d6fd27eb15b9d0bd99b
Success downloading: /airrohr/beta/latest_de.bin
Success downloading: /airrohr/beta/latest_de.bin.md5
Success downloading: /airrohr/loader-002.bin
launching 2nd stage
Restart.
Found firmware MD5: 40099cbc1cb8cf9aa792a0e7939584f4

Moving Firmware image to old.
Finished successfully.. Rebooting!

Airrohr: NRZ-2019-128-B6/DE
mounting FS...
opened config file...
parsed json...
Rewriting old config from: NRZ-2019-128-B7
Saving config...
Config written successfully.
output debug text to displays...
Connecting to xxx
.......
WiFi connected, IP is: x.x.x.x
Starting Webserver... x.x.x.x

ChipId: 3912888
Read SDS...: 16-11-21(063d)
Stopping SDS011...
Read DHT...
Read BMP280/BME280...
Trying BMP280/BME280 sensor on 76 ... found
Send to :
sensor.community
Madavi.de
----
Auto-Update active...
```

Er hat gleich ein Autoupdate gemacht und nun ist die NRZ-2019-128-B6 drauf. Spannend.

NTP hat er nun wohl auch bekommen. Ich wüsste auch nicht, was ihn bei mir im Netz daran hindern sollte, mit der NRZ-2019-128-B3 auf einen NTP Server im Internet zu kommen. Habe hier nichts eingeschränkt.

dirkmueller, 2020-01-03:
Danke. Damit sollte das urspruengliche Systemzeit problem nicht mehr auftreten. @msebald hattest du vorher mal hard resetted (stromlos) gemacht? Es gab keine relevanten aenderungen zwischen -B3 und -B7 was NTP in irgendeiner weise anders macht. wenn bei -B3 die systemzeit nicht gesetzt war, kann ja u.U. "mal" passieren wenn gerade eine stabilitaetsschwankung des internets war waehrend der ESP eingeschaltet wurde (ist ja udp und lossly). ist halt schon extrem unwahrscheinlich weil er drei mal drei verschiedene server versucht (die selber nochmal auf verschiedene server aufloesen). 

Man koennte also 

* wenn systemzeit bekannt falsch ist vor einem OTA versuch (der alle 24 stunden gemacht wird) nochmal NTP absyncen. Damit sollte das "OTA haengen bleiben" problem gefixed werden, und das ist eigentlich das wichtigste problem, weil sonst koennen wir sensoren nicht mehr remote fixen.

* nochmal weitergraben warum NTP nicht wie eigentlich dokumentiert einmal die stunde nachgefuehrt wird (das wuerde dann auch obigen punkt loesen)

msebald, 2020-01-03:
@dirkmueller Gerne. Das Device war zig Mal stromlos, da ich es ja mehrfach zwischen Netzteil und Notebook hin und her gesteckt habe. Auch jetzt als das Notebook in der Dachkammer irgendwann mal in Energiesparen ging.

Es klingt nicht so, dass etwas bei NTP schief gehen könnte - wenn die Internetleitung steht. Allerdings habe ich das Device ja so oft neu gestartet, soft wie hart, dass es ja irgendwann mal hätte klappen müssen. Und bei NRZ-2019-128-B7 klappte es dann ja direkt auf den ersten Versuch. Das sind schon Indizien, dass das Problem nicht bei meiner Internetverbindung zu suchen ist. Zumindest nicht grundsätzlich.

Warum habe ich nach der manuell geflashten NRZ-2019-128-B7 die NRZ-2019-128-B6 als Downgrade (statt Upgrade) erhalten?

Grundsätzlich: Mein Fokus liegt beim Feinstaubsensor schon primär auf Anwendung als auf Debugging, wobei ich hier natürlich gerne zur Verfügung stehe. Sollte ich daher wieder auf eine stable Firmware wechseln und falls ja, wann? Wenn die Betas nahezu so stabil laufen, bleibe ich aber auch gerne auf Beta. Aber wie gesagt, Fokus ist die Anwendung, also Messung der Daten. ;-)

dirkmueller, 2020-01-03:
@msebald klar, ich danke dir auch ganz herzlich für die Debugging Hilfe. Ich bevorzuge auch das daten sammeln, deswegen versuche ich fehler die leute vom updaten und stabilen sammeln abhalten immer zuerst zu finden.

An sich ist nichts einzuwenden gegen den beta channel. Ich habe den auch auf meinen produktiven sensoren subscribed. Er sollte absolut immer zum Datensammeln gehen. Es kann jedoch sein dass es probleme gibt die ein erfolgreiches OTA verhindern, wie hier passiert. Solange man also mal eben nen laptop anstecken kann um mit ner neuen Firmware zu recovern ist das okay. Auf stabil sollte das nie passieren, ist also der bessere default da dadurch der Sensor weniger händische Wartung braucht.

 Das mit dem "downgrade" ist normal. Wenn OTA an ist wird die Firmware beim starten und alle 24 Stunden mit was auch immer auf https://firmware.sensor.community/airrohr/ im jeweiligen channel liegt gesynced. Das erlaubt uns schnelles Roll-Back auf einen älteren aber funktionierenden Stand sollte es irgendwie ein größeres Problem mit einer neuen Version geben, solange OTA selber noch geht. -B7 ist noch nicht released.

dirkmueller, 2020-01-03:
Dass NTP nie ging mit der alten beta aber immer mit der neuen ist äußerst eigenartig, denn in dem Teilbereich hat sich vom code her nichts verändert.    Die eigentliche Ursache ist also nich da, hat aber gerade zufällig keinen Effekt mehr. 

Ich suche mal weiter..

msebald, 2020-01-04:
@dirkmueller Nochmals gerne und kein Problem. Mache ich gerne - trotz Fokus auf Datenerfassung (die ich in meinem Fall halt auch noch für die Hausautomation verwende - hier geht es v.a. um Temperatur und ein wenig um Luftfeuchtigkeit). Ich werde mal auf Beta bleiben und schauen, wie das nun weiter läuft. Klingt ja recht vielversprechend, was Du da schreibst.

Danke auch für die Info bezüglich dem "Downgrade". Das mit B7 wusste ich nicht. Daher war ich halt ein wenig verwundert und dachte, ich frage mal nach ob Bug oder Feature.

Mit NTP ist es schon seltsam. Aber ich würde fast auch die Wörter "immer" oder "nie" verwenden wollen. Immerhin habe ich zig Mal den Sensor hart neu gestartet weil stromlos gemacht und es klappte nicht. Bis ich die neuere Firmware drauf gemacht hatte und dann klappte es halt direkt. Schon komisch.

Wie machen wir's denn mit den oben besprochenen Features, die ggf ganz sinnvoll wären. Also Netzwerkconfig über Web einsehbar und auch den seriellen Output sichtbar über die Webseite? Neues Issue hier damit es nicht untergeht oder ist das schon notiert oder abgelehnt?

dirkmueller, 2020-01-05:
NTP debuginfo ist bereits implementiert in der aktuellen beta. fuer das console thema habe ich ein #612  aufgemacht. 

Mir ist noch aufgefallen dass es eine funktion gibt um die SDK config area zu wipen. ich hoffe mal das macht unser krams ausreichend stabil nach OTA.

commit fbdd92ed (2020-01-05): Erase SDK config on OTA restart (Fixes #590)

We need to wipe SDK config (twice) just before restart to
ensure that SDK version mismatches between 2nd stage
and old or new airRohr version are not causing hangs
or wifi dropouts.
closed 2020-01-05 by dirkmueller