opendata-stuttgart/sensors-software #597: Add dehumidifier HECA from Nettigo support

DeeKey, 2019-12-28:
Polish company Nettigo made an open source design to create a humidifier.
https://nettigo.eu/products/1709

They also added support for humidifier in their branch of code:
https://github.com/nettigo/air-monitor/tree/master/airrohr-firmware

Also we need to add (translate) a proper documentation how to assemble this thing.

dirkmueller, 2019-12-30:
as far as I can see the HETICA support in that fork of the firmware is simply addition of I2C SHT30 sensor.

dirkmueller, 2019-12-30:
I agree the documentation should be integrated, it seems a useful feature..

DeeKey, 2019-12-30:
There is a code to control HECA.
https://github.com/nettigo/air-monitor/blob/d9b0e976d07870d9ac02b351f050090dd9faf779/airrohr-firmware/airrohr-firmware.ino#L3641

Also from description:
"In this design we used precise temperature and humidity sensor from Sensirion SHT30 connected via I2C. Sensor is equipped with ALERT output. This output is connected to N-channel MOSFET and controls PTC Heating Element. Huge advantage of this design is ALERT logic which can be programmed via I2C and it's independent from main micro controller. In other words apart of sensor initialization it doesn't require any action from main micro controller."
https://nettigo.eu/products/nettigo-air-monitor-heca-ptc-heating-element-control-assembly