opendata-stuttgart/sensors-software #598: Unstable or no access to web interface on certain type of ESP8266

DeeKey, 2019-12-29:
Got as far as I understand several NodeMCU v2 boards.
https://frightanic.com/iot/comparison-of-esp8266-nodemcu-development-boards/#official-vs-unofficial

After load of /builds-NRZ-2019-128-B3-SDS011-fix-long-delay binary I can access WEB interface and enter AP credentials.
But after that I cannot access WEB interface and even ping device
Though few times I managed to access web interface.

Here is what happens after restart:

`2dO⸮4⸮$0z⸮4C8I⸮⸮Airrohr: NRZ-2019-128-B3/EN
`mounting FS...
`opened config file...
`parsed json...
`output debug text to displays...
`Connecting to ap
`.......
`WiFi connected, IP is: 192.168.1.127
`Starting Webserver... 192.168.1.127
`
`ChipId: 6519863
`Send to :
`sensor.community
`Madavi.de
`----
`SNTP sync finished: Sun Dec 29 11:52:03 2019

Added initial log (start after load of binary version)
[initlog.txt](https://github.com/opendata-stuttgart/sensors-software/files/4008005/initlog.txt)

Photos of the board:
[image]
[image]

dirkmueller, 2019-12-29:
Thanks for sharing all the details here! very useful. what I can see is that the initlog looks good. What I can also see in this board variant is that it doesn't have the metal shield around the esp8266+flash chip like "normal" esp8266 boards have (you can see it shine silver/golden here: https://dziadalnfpolx.cloudfront.net/blog/wp-content/uploads/2015/09/official-nodemcu-development-board.jpg )  instead flash and esp are blank. 

The typical concern here is that the wifi sending is causing electromagnetic interference with the flash reading and might trigger instability. so such boards are generally a very bad idea. 

What we can try is build a firmware that has lower WiFi sending power.

dirkmueller, 2019-12-29:
please try the firmware from https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B4-no-wifisetup/

DeeKey, 2019-12-30:
Just tested - it works! 
Here is an initlog
[initlog.txt](https://github.com/opendata-stuttgart/sensors-software/files/4009423/initlog.txt)

Also was possible to make an autoupdate. After update it stopped working, as expected (because of version downgrade).
[autoupdate.log](https://github.com/opendata-stuttgart/sensors-software/files/4009426/autoupdate.log)

Even though initial problem is solved we steel need to document this issue to warn users not to buy such ESP8266 anymore. 
Also not clear  if SW can sort this problem itself? Or we need to add a separate option to lower WIFI transmission power?

dirkmueller, 2019-12-30:
Thanks for testing. can you please also try out these two firmwares and let me know which one of them is working as well?

https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B4-persistent-true/
https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B4-persistent-true-no-output-power/

dirkmueller, 2019-12-30:
Yes, what I found is that reportedly lowering the Tx power solves "this" stability issue (still not entirely sure which of the settings that were removed made it work for you). I do wonder though if we have a way to detect that situation and do it automatically..

dirkmueller, 2019-12-30:
we might also hit https://github.com/esp8266/Arduino/issues/6886 here..

DeeKey, 2019-12-30:
Both version are working. Though I did not test stability.

With build NRZ-2019-128-B4-persistent-true everything works as expected
WiFi | signal strength | -47 dBm
WiFi | signal quality | 100 %
Device was close to router.
Log:
[builds-NRZ-2019-128-B4-persistent-true.log](https://github.com/opendata-stuttgart/sensors-software/files/4010475/builds-NRZ-2019-128-B4-persistent-true.log)

With build NRZ-2019-128-B4-persistent-true-no-output-power works as expected
WiFi | signal strength | -54 dBm (but later it became 45 dBm)
WiFi | signal quality  | 92 %

Log:
[NRZ-2019-128-B4-persistent-true-no-output-power.log](https://github.com/opendata-stuttgart/sensors-software/files/4010682/NRZ-2019-128-B4-persistent-true-no-output-power.log)

dirkmueller, 2019-12-30:
thats really confusing.

dirkmueller, 2019-12-30:
@DeeKey does the firmware from https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B4-no-wifi-off/ work?

dirkmueller, 2019-12-30:
its possible that the issue only reappears after ~ 30 minutes? not sure how long it took for you to reproduce the issue before?

DeeKey, 2019-12-30:
With stable build the problem appears straight away. When I was making an autoupdate then right after the stable version was updated the WEB interface stopped working.

Also, may I ask you to distinguish builds you are making for tests so that it can be clearly seen from the log. Otherwise there is a chance to mix versions...

dirkmueller, 2019-12-30:
@DeeKey I prepared a build that identifies itself as B5:

https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B5-gitd1b45d02/

If this work we're good.

DeeKey, 2019-12-31:
https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B4-no-wifi-off/ 

Yes, this build works. 
The only strange thing is that the menu 
http://192.168.x.xxx/status do not load
It tries to load it but fails after a long time.

DeeKey, 2019-12-31:
https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B5-gitd1b45d02/
This build also works but still the same problem. The status page does not load. At the end I got browser error "The connection was reset"

dirkmueller, 2019-12-31:
That likely means it crashed while rendering the page. Is that in AP mode or in sta mode?

Can you capture the serial output while accessing that page?

DeeKey, 2019-12-31:
Yes it is crashed. Will make another log on another ESP8266 of this type.
[status_page.log](https://github.com/opendata-stuttgart/sensors-software/files/4012354/status_page.log)

DeeKey, 2019-12-31:
That is in sta mode. Status page is not available in AP mode.

Also the same problem with another same type ESP8266
[status_page_2.log](https://github.com/opendata-stuttgart/sensors-software/files/4012434/status_page_2.log)

dirkmueller, 2019-12-31:
thanks, fixed!

DeeKey, 2019-12-31:
Any build to test?
A long run test should definitely be performed...

dirkmueller, 2019-12-31:
https://static.dmllr.de/airrohr/beta/builds-NRZ-2019-128-B6/

DeeKey, 2020-01-01:
works as expected!

closed 2020-01-01 by DeeKey