opendata-stuttgart/sensors-software #599: AntiFog hardware system

DeeKey, 2019-12-30:
Fog is one of the main source of uncertainty for particle meters. SDS011 will start showing high results when it is over 70% of relative humidity.
This problem is one of the main source of criticism for the project - "Your devices cannot distinguish fog from pollution. HAHA."
Some environmental criminals use the fog factor to hide their illegal combustion without being noticed by activists who monitor this with the help of PM sensors.

Yet, one must understand that the fog in polluted areas might be even worse then just particles. Water drops might accumulate hazardous chemicals and particles inside and thus pose even more problems to health and environment!

So in order to filter fog factor some AntiFog system must be used with particle meter sensor.
It might be very useful and interesting to compare values with and without AntiFmog system.

The perfect AntiFog system must be durable, affordable and easy to construct.
Currently I know only one ready made AntiFog system. This is HECA kit
https://nettigo.eu/products/nettigo-air-monitor-heca-ptc-heating-element-control-assembly

Also there are other initiatives by individuals which are not yeat ready for mass implementation:
https://github.com/opendata-stuttgart/sensors-software/issues/333

I created this issue to solve the problem of fog uncertainty. Please help to solve this issue!

gvidinski, 2020-02-12:
I'm also interesting in solving this issue. Living in an area with high thermal inversion weather activity we have a lot of foggy days during the winter. From what I can see on the map it is a general problem for most of the stations running around. When the fog comes almost every sensor reports humidity above 90%. And as all we know from experience such humidity affects the readings of the SDS011 (probably other particle sensors would be affected too if no counter measures are taken). So, lets create some road map. Starting from What we have, What we need, How to achieve at least PoC state. Recently I'm working on improved Solar radiation shield for this setup with the optional photo-voltaic/battery power. And because it is modular I'll be happy to give this HECA a mechanical test.

ricki-z, 2020-02-13:
Before you start such a project: We got samples of a new sensor that would have such a heating included. At the moment we are working on supporting this sensor in our firmware.
You can find more infos here: https://next-pm.fr/

gvidinski, 2020-02-13:
> Before you start such a project: We got samples of a new sensor that would have such a heating included. At the moment we are working on supporting this sensor in our firmware.
> You can find more infos here: https://next-pm.fr/

Sounds great. But I have a few questions.
Can you please tell us from where this sensor could be ordered? I'm unable to find any distributor of the NextPM sensor. 
What price has gave to you the manufacturer and how do you plan to distribute it around the world? 
From your experience how long it will take to implement this sensor? 
How easy it will be for the participants to put hands on it? Still you have to guarantee that all parts for the Base station project are easy to find in the light of DIY meaning. 

In such situation I don't see anything wrong in Researching and Developing of implementation additional module such AntiFog that could only be in great help.

ricki-z, 2020-02-13:
It seems like RS Components will distribute it: https://de.rs-online.com/web/p/luftgutesensoren/1953770
Prices are much higher than for a SDS011 or a SPS30. But this could be used even by inexperienced users (only one cable to connect). The HECA kit needs assempling and soldering, so many users may not be able to build this. Searching a manufacturer who can assemble this may not be much cheaper than the mentioned price.

gvidinski, 2020-02-13:
Another great news! 
But what about all those already assembled stations, which uses the old SDS011, and all those users that had have already invested money and time to build them? Don't take me wrong. I'm trying to get the picture from as many view points as I can. Don't forget also those user that will love to use soldering iron and help others as well. May be it is still a good idea if we can provide those users a solution to improve their station readings, not having the need to replace the sensors at all. 
As I remember service life time of SDS011 is 8000 hours (by specs). This means that station configured to take readings on every 300 seconds, with sensor switched ON for 30 seconds (not sure but I think this time period is even less in latest firmware) could prolong the life of the sensor to over 9 years. In theory of course. I think it's worth it.

Adorfer, 2020-02-14:
Datasheet for "NEXT-PM" states "up to 95% non condensing". 
That's certainly not enough for the conditions when there is real fog outside. 
So even if that sensor is a great improvement, there will be need for an additional heating to drop at least from "fully saturated 100% with droplets in the air" down to 95%.

zdroyer, 2020-04-07:
It seems there is an interesting low cost solution for the high humidity. A Polish firm Nettigo released an OSH pcb based on SHT30 for preheating the air for SDS011, until the relative humidity goes below 70%.
 Here it is: https://nettigo.eu/products/nettigo-air-monitor-heca-ptc-heating-element-control-assembly 
They also created a full retrofit preheating chamber to be added to an existing luftdaten node: the heater, pieces of pipes and the PCB for ~18€ https://nettigo.eu/products/nettigo-air-monitor-heca-kit-heating-chamber-with-control-module
I read the documentation carefully: you can use the preheating device completely independently form the microcontroller (SHT30 hat the programmable ALARM function to control the heater), but you could also monitor the state of the heating chamber by connecting the pcb to the I2C bus of the microcontroller (the chamber SHT30 has customised address 0x44, so it doesn't collide with the default one, if it's already connected to the microcontroller).
I know Nettigo forked the luftdaten firmware to include the gather data from their preheating chamber: https://github.com/nettigo/air-monitor