opendata-stuttgart/sensors-software #600: High values visual indication with sound alarm or LED

DeeKey, 2019-12-30:
There is a feature request to indicate value with RGB led
# #149 
Solution with RGB requires some additional hardware and free analog pins, which we do not have.
What I propose is to indicate high values with alarm sound or one LED indication driven by a spare digital pin (hope there are still some free).
The idea is to give visual or sound indication so that people around can see it without any additional devices.

ropeters68, 2019-12-30:
Would this work (1 pin)? 
https://a.aliexpress.com/9AAGhtubv

ricki-z, 2019-12-30:
Normally the sensor will be positioned outside (and normally outside users view). So that is not really something we should do.
There are solutions like https://revspace.nl/StofAnanas that use another ESP8266 to read the values of the sensor and show appropriate colors.

DeeKey, 2019-12-31:
This can be used instead of GPS
D5, D6: PWM input PPD42NS, second software serial for GPS
D7: DHT22
As far as I understand D8 is totally free.

To add LED is just a resistor and the LED.  Such an option might be useful for private house or some public space where it is visible to others. I am not sure if separate ESD8266 needed.
But lets evaluate this issue. I will ask other people opinion on this feature request.

dirkmueller, 2020-01-02:
so if I understand this correctly, the LED should light up if the air quality is below (worse than) some air quality index level? I assume we're not going for the "sound" option because then we'd need some kind of "mute/acknowledge" interaction as well. 

I recently started implementing EAQI calculation and events reporting to IFTTT (maybe home assistant). would this be something good for you as well? Why only toggle a small LED when you can change the color temperature of the complete house lighting :-)

DeeKey, 2020-01-03:
> so if I understand this correctly, the LED should light up if the air quality is below (worse than) some air quality index level? 

True. The idea is to have some simple way of notification about air quality. For in the box installation one LED could be quite informative. This way you will not need to check your phone when going to ventilate the room.
Also HW part is quite simple - one resistor and LED. 

https://thumb.tildacdn.com/tild3837-6132-4736-b833-326332373338/-/resize/460x/-/format/webp/noroot.png

Home assistant, IFTT and other of this kind are more advanced solutions and hard to use for ordinal people without tech background...