opendata-stuttgart/sensors-software #605: IPv6 Support

genofire, 2020-01-02:
We run already a ipv6only network (with dns64 and nat64 - for ipv4only services in the internet).

dirkmueller, 2020-01-02:
First step would be to compile in IPv6 support (which is currently disabled due to size constraints)

genofire, 2020-01-10:
Do you have a hint, where i could find the build flags to try it on my own?

dirkmueller, 2020-01-10:
In the platformio.ini you find the lwip variant as a -D define. We currently only build with ipv4.

I tried getting ipv6 set up however my provider does not support it..

See https://docs.platformio.org/en/latest/platforms/espressif8266.html#lwip-variant for the possible build flags

dirkmueller, 2020-01-12:
Ist nun in B2 so gesetzt. Getestet habe ich noch nichts

genofire, 2020-01-12:
German ^^: nice, werde ich am Dienstag ausprobieren xD

genofire, 2020-01-16:
So, er ist nun im DualStack Netzwerk erreichbar :+1:  im IPv6only Netz keine Ahnung.

Frag mich gerade, wie man ggf. die Software etwas härten könnte:
- HTTPS (selfsigning oder Let's Encrypt)
- IP-Subnet white-/blacklist als Firewall oder im "Webservice"

dirkmueller, 2020-01-16:
@genofire yeah, there are security implications. I am not a big fan of https webserver right now because the way Arduino core implements this is with very high overhead. I have lost a bit of confidence on being able to fix that quickly.

We likely need to restrict to link local addresses for now, everything else is too dangerous. Does the mdns/dns-sd resolve to ipv6? To ipv4? Both?

genofire, 2020-01-16:
please do not restrict to lladdr - so it's not reachable by e.g. firefox or other tools.
maybe restrict to:
- lladdr and given subnet 
- or a whitelist
- or let it like current state (most router like AVM and so on has a good firewall for such things) 

--- i do not use this features, but it will try mdns / avahi next thuesday

dirkmueller, 2020-01-16:
Okay. Does the user have to configure the whitelist prefix or can we just learn it via slaac?

I think we don't have enough insights on how sensors are being deployed to know whether we're safely able to fix on good filtering at the perimeter.

Given the gaping holes in the UI (Don't want to be more explicit until fixed) we do need to be careful with this.

Please advice what the minimum steps are to make this safe. If I can't get that implemented timely I will have to turn off ipv6 for a stable release until we had time to do basic things right.

genofire, 2020-01-17:
for the first release i would learn it by "route advertisement" (RA) (which is used as subnet of configurated by slaac) and maybe a checkbox to disable this firewall.

in 2. step a textarea in the UI will be nice, where the RA subnet is already in in format like:
```
fe80::/64
2001:db8:1:2::/64
```
---
but like i say, that is not a blocking think (a ipv6 network scan to find such devices, is not easy - on most private networks impossible thanks "good" routers firewall like the fritzbox)