opendata-stuttgart/sensors-software #607: Einstellbarer Temperatur-Offset

honkmaster, 2020-01-02:
Ich habe mein Airrohr in einer [Verteilerdose](https://www.thingiverse.com/thing:2775946) verbaut. In dieser Dose befindet sich auch der Temperatursensor (DHT22). Die Temperatur in dieser Dose ist im Vergleich zum offenen Design und externen Messungen drei Grad höher. 

Daher mein Feature-Request: Ich würde gerne einen Offset in der Firmware einstellen können, welcher von der gemessenen Temperatur abgezogen / hinzugefügt wird.

dirkmueller, 2020-01-03:
ja, die frage ist ob es wirklich nur ein linearer offset ist.. temperatur koeffizienten haben die tendenz zu nichtlinearitaet, je nach temperaturdifferenz zur waermequelle und der waermeleitfaehigkeit (luftfeuchte!!)

und ich kann mir nicht vorstellen dass user korrekt polynome dritten grades mit den koeffizienten bestimmen koennen..

honkmaster, 2020-01-03:
Ich habe es über ein Jahr beobachtet. Der Unterschied beträgt 2-3 Grad. Mir würde ein Offset genügen (Nichtlinearität, Polynom, ..., Warum immer gleich so kompliziert denken). Warum den Offset nicht irgendwo in Debug-/Entwickler-Settings verstecken. Muss ja nicht für jeden sofort sichtbar sein.

dirkmueller, 2020-01-12:
Ist nun in der aktuell releasten beta enthalten (NRZ-2020-130-B2)

closed 2020-01-12 by dirkmueller

honkmaster, 2020-01-12:
Danke @dirkmueller 👏

honkmaster, 2020-01-12:
Eine Anmerkung: Ich würde den "Korrekturwert in °C:" in den Einstellung in den Tab Sensoren unter den "Korrekturwert in dB(A)" verschieben. Dort passt er besser hin.

dirkmueller, 2020-01-12:
Done

honkmaster, 2020-01-17:
2 Anmerkungen: 
- Ich habe einen BME280 nachgerüstet, dieser hat einen anderen Offset als der DHT22...brauch man vielleicht einen Offset je Sensor?
- Die Firmware NRZ-2020-130-B3 wird nicht installiert.

commit 050e4cbf (2020-01-12): Add configurable temperature offset (Fixes #607)

This allows to tune the temperature reading by a fixed offset.