opendata-stuttgart/sensors-software #610: Suport for the insplorion NOx sensor

JanBosNL, 2020-01-04:
Would be great if we could have a budget low energy NOx sensor, and I am curious if anybody knows this swedish manufacturer and has a spec sheet to see if this sensor might be an addition for measuring open air nitrogen levels in a useable and trustworthy way.

https://www.insplorion.com/en/air-quality-sensing/

ricki-z, 2020-01-05:
Hello,
we are in contact with this company. The sensor is in beta at the moment . The date of the final version is jet unknown. We hope to get some of the first examples of the final version in Q3/Q4.

JanBosNL, 2020-01-05:
@ricki-z Thats great to hear, sounds like a great addition for the luftdaten sensor module. Now lets al hope its within a diy budget range. For me the only other option would be the Alphasense NO2-B43F but the cost of 200€ is not within my range. And we truly need a cheap and trustworthy NOx sensor. especially in the Netherlands where acid rain and NOx polution is truly a big problem. 

And it seems the state is not even trying to get a decent country wide measurement because of political reasons. I believe the idear is what people cant see is no problem. There are big areas on the netherlands NOx data map that are blind spots. Where no data is collected. Especially around Brabant and Friesland. (Agricultural areas)

see the map https://www.luchtmeetnet.nl/stations/alle-provincies/alle-gemeentes/NO2

We need this so badly!

Thanks for the reply and good news!

dirkmueller, 2020-01-15:
@JanBosNL please also take a look at #628

JanBosNL, 2020-01-16:
Looks promising, a lot depends on quality and lifespan. If the price mentioned is correct then this would be a realy nice adition.