opendata-stuttgart/sensors-software #612: Capture serial output in a small ring buffer for web console

dirkmueller, 2020-01-05:
For debugging sensors without having to carry laptops etc to difficult to reach (or cold :-(( ) locations it would be useful to have the last 512 bytes of serial output available in the /status webpage. 

probably we just need to create a small wrapper that overloads the virtual write() method and stores it (also) in a memory backed ring buffer.

DeeKey, 2020-01-05:
It would be nice to automatically send crash reports Like Linux distros do.
But that will be the next step I guess...

dirkmueller, 2020-01-15:
The serial debug mirroring has been implemented in a slightly hacky way. Will consider this issue done even though it will still need  a bit of polishing going forward.

commit b9963e4a (2020-01-12): Show a serial ring buffer on the webui (Fixes #612)

This allows debugging / looking at serial output
without actually using serial output.
closed 2020-01-15 by dirkmueller