opendata-stuttgart/sensors-software #615: airrohr-firmware fails to connect to wifi

Qwertie-, 2020-01-07:
I have configured everything in the same way as [the manual](https://luftdaten.info/en/construction-manual/#komponenten-schaltung). After flashing the firmware I can get to the config page to enter the wifi details and then the page reloads and is blank. After that the device never connects to the wifi AP I just entered. Tried this a few times with the beta and non beta firmware a few times but had no luck.

dirkmueller, 2020-01-07:
can you connect the node to  usb-serial debug https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/Readme.md#debug-via-usb-serial and paste its output when it tries to connect to wifi?

dirkmueller, 2020-01-07:
is there something special about your access point? note that this hardware requires 802.11n on 2.4Ghz to be allowed by the AP. 5Ghz (which is the modern default these days) does not work.

Qwertie-, 2020-01-07:
I'll give that debug tool a test tonight. I don't think there is anything special about the setup. I attempted to connect to my rotuer 2.4ghz and a hotspot from my phone. The networks show up on the config page and I clicked on it to fill the network name.

dirkmueller, 2020-01-07:
Thanks, that sounds promising. Other guess would have been that it uses one of the exotic wifi channels (channel 12+) and it for some reason refuses to use those.

Qwertie-, 2020-01-08:
I'm seeing this on a loop
[image]

dirkmueller, 2020-01-08:
so this is NRZ-2019-124-B10. with other words you're one of the two that are still on this old version.. can you please flash the latest one, downloaded from here: 

https://firmware.sensor.community/airrohr/update/ (choose latest_XX.bin where XX is the locale you're interested in). 

using this command line:

```shell
esptool.py  --chip auto --port $dev --baud 460800 write_flash -fm dio 0x00000  latest_xx.bin
```

Qwertie-, 2020-01-08:
```
Connecting to OpenWrt
...........
WiFi connected,
```

:+1: 

Any idea why https://github.com/opendata-stuttgart/airrohr-firmware-flasher/ includes an old version of the firmware?

dirkmueller, 2020-01-08:
I've never used this software, but it seems it simply refers to the abandoned firmware server. we switched to sensor.community  a few months ago. 

https://github.com/opendata-stuttgart/airrohr-firmware-flasher/pull/23 should fix this.

dirkmueller, 2020-01-08:
Thanks for the help troubleshooting this!

closed 2020-01-08 by dirkmueller